FreeOpcUa/python-opcua #175: ModifyMonitoredItemResponse broken?

oroulet, 2016-05-03:
Wireshark says we send a malformed packet in response to an ModifyMonitoredItemRequest for event subscription.
This is not sure we have an issue, wireshark is sometime wrong and our code is autogenerated and is usually correct.

destogl, 2016-05-03:
I will try to subscribe on events from unifiedautomation SDK server so we will see...

destogl, 2016-05-03:
How do you get ModifyMonitoredItemRequest?

oroulet, 2016-05-03:
to get a modify monitored item requtes you need to modify the subscription by clicking on attributes of subscribed objects in event view

Johen8, 2016-05-11:
Yeah, I think it is broken. When using UaExpert to connect to a server I get this message when trying to Modify a Monitored Item:
[image]

Then I implemented on the client side a function to Modify Monitored Items (I couldn't find one, at least) and I get the same message when trying to use it:
`[MonitoredItemModifyResult(StatusCode:StatusCode(Good), RevisedSamplingInterval:0.0, RevisedQueueSize:1084178432, FilterResult:None)]`

After this, I tried to use the same function with the UAF server and I got the following message, using the same parameters and function from the client:
`[MonitoredItemModifyResult(StatusCode:StatusCode(Good), RevisedSamplingInterval:5000.0, RevisedQueueSize:1, FilterResult:None)]`

So, I guess the problem is in the server.

(I couldn't get my modify monitored item to function properly after the modification, I get this error:
`WARNING:opcua.common.subscription:Received a notification for unknown handle: 202`
If anyone could help me, I'd appreciate it)

closed 2016-06-12 by oroulet