FreeOpcUa/python-opcua #178: Monitor Item with DeadBandValue?

Johen8, 2016-05-04:
So, I was looking into how I could measure changes in a variable with some noise and in the documentation for OPC I found the DeadBandValue option for monitoring changes in a variable, since the simple 'datachange' option that is implemented isn't good enough for this. I looked into the code and I found that there is a class implemented for this, but in the auto generated code only, it seems. I could not find any implementation for this kind of filter for monitored items, or did I miss something?

oroulet, 2016-05-04:
Server or client?
Support for deadband is not implemented in server.
On client side you can create a monitoredItem by hand and specify this parameter. But good luck, when you can understand this, you will be ready to contribute to python-opcua :-)

zerox1212, 2016-05-04:
@oroulet Sorry for asking, but could you take a few minute to make a "custom-MonitoredItem" example. You have spoke about this a few times, but I think most people will not know where to start on this.

oroulet, 2016-05-04:
If I get two minutes I will do. But simply look at Subscription class. It has a method called create_monitored_item taking a MonitoredItem as argument. If you look at code of  subscribe_event method you should manage to understand how to create a MonitoredItem.

Johen8, 2016-05-04:
I'm going to use the code on a project at my University. I started studying it and the OPC specifications this monday, and as soon as I feel capable, I will definitely start contributing =D

oroulet, 2016-05-04:
Great. Hading support for deadband in server should not be hard

zerox1212, 2016-06-03:
Can we close this now?

closed 2016-06-12 by oroulet