FreeOpcUa/python-opcua #191: Client alarm functionality

drzippiz, 2016-05-27:
Sorry if this is a simple question, but I noticed that alarms are neither listed as supported or not supported for the client code. I don't see any functions directly relating to alarm use, but is it possible to monitor an alarm as an event? Or is this functionality just not present?

oroulet, 2016-05-31:
alarm are not supported yet, but it looks like we get alarms as events.....try and le us know

zerox1212, 2016-06-03:
We just finished events recently and now a requests for alarms lol.

I can look into adding alarms soon, or maybe you can help us out @drzippiz ?

georgschoelly, 2016-07-25:
I'd be willing to invest a bit of time to implement basic alarm functionality for the client. Did you have a specific approach in mind?

oroulet, 2016-07-25:
I do not know anything about alarms. so just try and try to make an API similar to the existing. One thing I would suggest is to create very early a pull request (as soon as you have some code) and name it something like "Alarm DO NOT MERGE" so people can look early at what you do and comment. This will probably save you a lot of time. Then you can create later a cleaner PR.