FreeOpcUa/python-opcua #192: Empty certificate

nuccy, 2016-05-30:
Hello.
Recently I tried to create the OpcUA server to be connected via proprietary software, which uses Prosys OpcUA Java SDK and I've experienced the problem at the very beginning of the connection:

Prosys client reports: 
Exception in thread "AWT-EventQueue-0" org.opcfoundation.ua.common.RuntimeServiceResultException: org.opcfoundation.ua.common.ServiceResultException: Bad_CertificateInvalid (code=0x80120000, description="2148663296, Could not parse certificate: java.io.IOException: Empty input")
(See more in the attached file) [prosys_client.txt](https://github.com/FreeOpcUa/python-opcua/files/289196/prosys_client.txt)

During that the examples/server-minimal.py reports simple lost of connection and:
('Cleanup client connection: ', ('127.0.0.1', 47298))

On the Prosys forum they have similar issue: http://forum.prosysopc.com/forum/opc-ua-java-sdk/certificate-problems/ but they suggest that the problem is in the OpcUA server, which simply  returns an empty certificate string, rather than null. 

Please instruct how to fix that, because if I specify any certificate via server.load_certificate(_ANYTHING_HERE_) the encryption is activated which I simply don't need as far as proprietary software was created to contact unencrypted OpcUA servers.

nuccy, 2016-05-30:
P.S. I can supply more information if needed either from server side or from client side.

oroulet, 2016-05-30:
yes python-opcua defaults string and bytes to empty instead of None.  We should change this. Here is a pull request. It solve the login issue with prosys client. but we should check this a bit more before merging. Can you try it? https://github.com/FreeOpcUa/python-opcua/pull/193

nuccy, 2016-05-30:
Hello oroulet,
Thanks for really fast reply. I've tested your changes, indeed now Prosys SDK Java Client connects to python-opcua without any problems and seems to work fine.

oroulet, 2016-05-30:
thanks. I merged the changed to master. I also did some tests and this does not seem to break anything ...yet...

closed 2016-05-30 by nuccy