FreeOpcUa/python-opcua #195: Methods' parameters

nuccy, 2016-05-30:
Hello.
While using proprietary Prosys opc-ua client, which uses Prosys OpcUa Java SDK, I've found that the client incorrectly lists the types of method's arguments, e.g. when starting the python-opcua server from example/server-methods.py, the client cannot call any of two available methods responding with an error: Error in getting input argumentsjava.lang.NullPointerException
(full error message in the attached file) 
[prosys_list_method_parameters.txt](https://github.com/FreeOpcUa/python-opcua/files/289722/prosys_list_method_parameters.txt)

Moreover it is even not able to show a form where these values of these arguments can be set. Not sure that this is the problem of python-opcua, but when using some proprietary OpcUA server, which is running inside PLC the methods can be called correctly.

destogl, 2016-05-31:
Hi nuccy, we tested the server with UaExpert which is also showing the fields and there seams everything OK. Could you compare messages using wireshark with working server and python-opcua server?

oroulet, 2016-05-31:
prosys has a record of being good at following the spec so I installed their client and did some tests. I found that we had the wrong data type for some attributes, fixing this seems to also fix the method call stuff. Update to master it should work. thanks you for reporting this!

nuccy, 2016-05-31:
Hello destogl, oroulet,
I confirm that now it works as expected, thanks!

closed 2016-05-31 by nuccy