FreeOpcUa/python-opcua #196: NodeID

nuccy, 2016-05-31:
Hello again,
Not sure that this is an issue (if not please point to manual how to change that) the NodeIDs which are assigned to variables, methods or properties look like ns=2;i=2004 (e.g. for myarrayvar in examples/server-methods.py), but in other opcua servers (mostly proprietary) they have the NodeIDs e.g. "ns=2;s=Application.PLCPROG.Unlock" which is a bit more human readable as far as Application is the NamespaceURI, PLCPROG is a folder and Unlock is the variable. I'm asking that because I need to put around 1300 indexed variables (like pixel0001, pixel0002, etc.) into opcua server and it would be much easier to access them via 'ns=2;s=App.Camera.pixel%s' % id, than ns=2;i=XXXX. Thanks.

nuccy, 2016-05-31:
P.S. Sorry, In the string "ns=2;s=Application.PLCPROG.Unlock" the "Application" is also a folder, ns=2 is responsible for the Namespace.

Johen8, 2016-05-31:
Yes, you can do that. This is only the numeric ID of the node. Take a look at the server-example.py

You have this on the server:
```python
myobj = objects.add_object(idx, "MyObject")
myvar = myobj.add_variable(idx, "MyVariable", 6.7)
```
And on the client, you can access this variable through:
```python
myvar = root.get_child(["0:Objects", "2:MyObject", "2:MyVariable"])
```
So you can create objects/variable with a loop with the name you want and also access them in the same manner.

oroulet, 2016-05-31:
You can do as Johen8 says but add_variable also can take a nodeid and
qualified name as argument so you can give explicit nodeids using string if
you want
Add_variable(NideId('myapp.myvar1',2), qualified name('myname', 3))

nuccy, 2016-05-31:
Johen8, ok, but I'm not using the python client, but the one which uses Prosys Java SDK, where I need to specify something like "ns=2;s=Application.PLCPROG.Unlock"

oroulet, your suggestion:
Add_variable(NideId('myapp.myvar1',2), qualified name('myname', 3))
seems to be incorrect, currently I use:
```
T = main_folder.add_object(idx, "T")
myvar = T.add_variable(idx, "11", 1.1)
```

P.S. Sorry for probably stupid questions but I'm new to your python-opcua and want to be fully complaint with what was already done on Java from client side.

oroulet, 2016-05-31:
My suggestion should work :-)

Johen8, 2016-05-31:
Did you correct the typo? It should be `NodeId` instead of `NideId`

nuccy, 2016-05-31:
:)
Sure,
`Add_variable(NideId('myapp.myvar1',2), qualified name('myname', 3))`
to 
`myvar = T.add_variable(NodeId('myapp.myvar1',2), QualifiedName('myname', 3))`

and I get:
```
Traceback (most recent call last):
  File "../../OpcUaCamera.py", line 81, in <module>
    myvar = T.add_variable(NodeId('myapp.myvar1',2), QualifiedName('myname', 3))
  File "/home/user/OpcUaCamera/python-opcua/opcua/common/node.py", line 449, in add_variable
    return manage_nodes.create_variable(*args, **kwargs)
  File "/home/user/OpcUaCamera/python-opcua/opcua/common/manage_nodes.py", line 71, in create_variable
    val = _to_variant(*args[2:])
TypeError: _to_variant() takes at least 1 argument (0 given)
```

oroulet, 2016-05-31:
Almost correct, you still need a default value at the end then it should
work.
Sorry many typo here, I write from my phone...

nuccy, 2016-05-31:
Thanks, indeed missed the last argument.

closed 2016-05-31 by nuccy

nuccy, 2016-05-31:
Just in case somebody will have the same issue and will find this issue, here is the final correct code:

```
main_folder = objects.add_folder(idx, "Camera")
T = main_folder.add_object(idx, "T")
for x in range(0, 12):
        myvar = T.add_variable(NodeId('Camera.T.pixel%d' % x, 2), '%d' % x, 0.0)
```

gomez100, 2016-08-31:
Hi nuccy,
I'm having the same issue but with your solution I'm receiving this error "NameError: name 'NodeId' is not defined". Can you provide me an example of working project" please? ( maybe i forgot some "import" ) Thanks in advance. Andrea

gomez100, 2016-08-31:
Hi,
solved with this syntax!
https://github.com/FreeOpcUa/python-opcua/issues/126