FreeOpcUa/python-opcua #197: autogenerated events

oroulet, 2016-06-01:
@destogl CustomeEvent class is flagged as auto-generated but I cannot find any script generating it. ??

destogl, 2016-06-01:
I flagged it since it was idea that we auto-generete it. in #159 you mentioned you wrote already some code for it. Do you have something? If not than we should write it...

oroulet, 2016-06-01:
OK. I was just wondering. Yes i started some code, I should check it in, even if it is not finished. I had to prioritize other things

georgschoelly, 2016-07-26:
@oroulet: That would be a big part of what I need for implementing a nicer interface for the alarms. (It already works using just the event functionality, but it's not nice to use and somewhat difficult to figure out.)

destogl, 2017-05-03:
Closed in #406

closed 2017-05-03 by destogl