FreeOpcUa/python-opcua #198: Unable to write values to kepware server by using python-opcua

nishantkumar5, 2016-06-02:
Hi,
Below is the setup:
- Kepware server running on my windows machine.
- Had set python-opcua.
- Able to read values from the server, however when trying to write a value through client it gives me below error:
opcua.common.uaerrors.UaStatusCodeError: The access level does not allow writing to the Node.(BadNotWritable)

- Same node however is writable through kepware client.

Setup contains:
Energy Meter, PLC, Industrial Flip switches.

Please see screenshots for more details.
Please suggest if i am going wrong. let me know if you need any further details.
[KepServerEX_OPCUA_WriteValue.docx](https://github.com/FreeOpcUa/python-opcua/files/295727/KepServerEX_OPCUA_WriteValue.docx)

oroulet, 2016-06-02:
Looks like you just different credential from the clients. Can you check?

zerox1212, 2016-06-03:
Does your Kepware server have security enabled?

nishantkumar5, 2016-06-03:
@oroulet - There is no credential setup at either KepserverEX or client.
@zerox1212 - I have cross checked and there is no security enabled onto kepware.

I am attaching screenshots for OPC UA configuration manager.
I am not sure if i am missing any configuration bit, your suggestions will help.
[OPCUA_ConfigManager.docx](https://github.com/FreeOpcUa/python-opcua/files/297032/OPCUA_ConfigManager.docx)

nishantkumar5, 2016-06-03:
Do you have any sample client program which writes values to remote server?

oroulet, 2016-06-03:
myfloat.set_value(0.0008, ua.VariantType.Float)

for example....
This is everywhere, by many people, so this definitely works. There must something special with your client. Have you tried to login as anonymous and write a node with uaexpert? Does that work?

Now I remember that kepware was really pircky with timestamps. try:

      dv = ua.DataValue(0.55, ua.VariantType.Float)
      dv.SourceTimestamp = None
      dv.ServerTimestamp = None
      myfloat.set_value(dv)

And try with different timestamp (datetimeobject) and different attribute of dv

nishantkumar5, 2016-06-03:
@oroulet - Thanks, i had corrected the syntax and it is working fine.

**Correct syntax:**
plc_out_01.set_value(0,ua.VariantType.Int16)

We can close this issue. Thanks a lot for your help!!!
[Client_writingtoKepServer_workingcode.docx](https://github.com/FreeOpcUa/python-opcua/files/297296/Client_writingtoKepServer_workingcode.docx)

closed 2016-06-03 by nishantkumar5

oroulet, 2016-06-03:
Well the conclusion is that kepware sends wrong error codes, one more issue with them. This program is rubbish