FreeOpcUa/python-opcua #201: High load problem

nuccy, 2016-06-06:
Hello, recently I've noticed that under pretty high load the python-opcua periodically throws errors:

```
ERROR:opcua.server.binary_server_asyncio:Exception raised while parsing message from client, closing
Traceback (most recent call last):
  File "/home/user/_dev/python-opcua/opcua/server/binary_server_asyncio.py", line 85, in _process_data
    ret = self.processor.process(hdr, buf)
  File "/home/user/_dev/python-opcua/opcua/server/uaprocessor.py", line 73, in process
    msg = self._connection.receive_from_header_and_body(header, body)
  File "/home/user/_dev/python-opcua/opcua/ua/uaprotocol_hand.py", line 592, in receive_from_header_and_body
    return self._receive(chunk)
  File "/home/user/_dev/python-opcua/opcua/ua/uaprotocol_hand.py", line 623, in _receive
    self._check_incoming_chunk(msg)
  File "/home/user/_dev/python-opcua/opcua/ua/uaprotocol_hand.py", line 550, in _check_incoming_chunk
    self._security_token.ChannelId))
UaError: Wrong channel id 113, expected 180
('Cleanup client connection: ', ('127.0.0.1', 43833))
```
```
ERROR:opcua.server.binary_server_asyncio:Exception raised while parsing message from client, closing
Traceback (most recent call last):
  File "/home/user/_dev/python-opcua/opcua/server/binary_server_asyncio.py", line 85, in _process_data
    ret = self.processor.process(hdr, buf)
  File "/home/user/_dev/python-opcua/opcua/server/uaprocessor.py", line 73, in process
    msg = self._connection.receive_from_header_and_body(header, body)
  File "/home/user/_dev/python-opcua/opcua/ua/uaprotocol_hand.py", line 592, in receive_from_header_and_body
    return self._receive(chunk)
  File "/home/user/_dev/python-opcua/opcua/ua/uaprotocol_hand.py", line 623, in _receive
    self._check_incoming_chunk(msg)
  File "/home/user/_dev/python-opcua/opcua/ua/uaprotocol_hand.py", line 550, in _check_incoming_chunk
    self._security_token.ChannelId))
UaError: Wrong channel id 113, expected 181
('Cleanup client connection: ', ('127.0.0.1', 43834))
```
```
ERROR:opcua.server.binary_server_asyncio:Exception raised while parsing message from client, closing
Traceback (most recent call last):
  File "/home/user/_dev/python-opcua/opcua/server/binary_server_asyncio.py", line 85, in _process_data
    ret = self.processor.process(hdr, buf)
  File "/home/user/_dev/python-opcua/opcua/server/uaprocessor.py", line 73, in process
    msg = self._connection.receive_from_header_and_body(header, body)
  File "/home/user/_dev/python-opcua/opcua/ua/uaprotocol_hand.py", line 592, in receive_from_header_and_body
    return self._receive(chunk)
  File "/home/user/_dev/python-opcua/opcua/ua/uaprotocol_hand.py", line 623, in _receive
    self._check_incoming_chunk(msg)
  File "/home/user/_dev/python-opcua/opcua/ua/uaprotocol_hand.py", line 550, in _check_incoming_chunk
    self._security_token.ChannelId))
UaError: Wrong channel id 113, expected 182
('Cleanup client connection: ', ('127.0.0.1', 43835))
```
After each such error the server shuts down and restarts.

Currently I have around 140 datapoint (mostly double) in 12 folders  and they are updated inside python script every 0.5 seconds. The client (Prosys Java SDK based client) queries all the datapoints around each second. What is the issue and how to solve it? Thanks.

oroulet, 2016-06-06:
Interesting... Maybe we have a lock issue looks similar to the long running traceback

nuccy, 2016-06-06:
The error occur irregularly (sometimes after hour, sometimes after minutes) while the inner updating and external opcua querying actions are stable (every 0.5 seconds and every 1 second respectively).

oroulet, 2016-06-12:
Does the entire server shut down? As I understand the code, only the client should get disconnected.

Can you try this patch https://github.com/FreeOpcUa/python-opcua/pull/206
IT will simply make the server ignore bad messages from client. This might be more correct than closing the connection, not sure...

nuccy, 2016-06-12:
Ok, will try and reply tomorrow because I don't have access to hardware today.

nuccy, 2016-06-13:
Nope, errors again, still the opcua server disconnects the client (see the attached file with errors
[python_opcua_errors.txt](https://github.com/FreeOpcUa/python-opcua/files/311522/python_opcua_errors.txt)
).

oroulet, 2016-06-13:
interesting, we do not get one but many messages with wrong tokenID. Kan you install wireshark and record a session? the file will get big, but maybe you can filter to only store opc ua message....

nuccy, 2016-06-13:
Yes, ok.
Recording the traffic with tcpdump (to wireshark readable file): 
`tcpdump -s 0 -w python_opcua.pcap -n -i lo dst port 4842 or src port 4842`
Is it fine for you?

nuccy, 2016-06-13:
So here are the errors 
[python_opcua_errors_while_dumping.txt](https://github.com/FreeOpcUa/python-opcua/files/311578/python_opcua_errors_while_dumping.txt)
The dump I will send in few minutes to your email.

oroulet, 2016-06-13:
how many clients? and what did they do?

nuccy, 2016-06-13:
There is one client (proprietary software running embedded opcua client based on Prosys OpcUA Java SDK), how many actual connections is hard to say because the software is not opensource. The client gets the values of all datapoints by querying ns=2;s=.....  addresses which are predefined in its configuration file.

Python opcua server is integrated inside software which gets data from canbus and every 0.5 seconds updates the datapoints by calling set_value()

nuccy, 2016-06-13:
The errors occurred  when the Prosys client was gettting the values of datapoints, usually it does than successfully but occasionally errors in python opcua server occur which were listed above.

oroulet, 2016-06-13:
your client get a new TokenId but still sends request using the old one. I check the spec and it seems to be allowed to use old TokenIds........maybe we should not test for that kind of things....

destogl, 2016-06-13:
Or maybe we shpuld test regarding a list of old and new tokenIDs... Eventhhough... This can get big and affect performance.

oroulet, 2016-06-15:
@nuccy Can you try again? https://github.com/FreeOpcUa/python-opcua/pull/206
Now we accept older ids... in theory, I have no client to test that kind of things

nuccy, 2016-06-15:
Ok, just started the server. Usually in an hour in disconnects once or twice, waiting.

nuccy, 2016-06-15:
Same again:
```
ERROR:opcua.server.binary_server_asyncio:Exception raised while parsing message from client, closing
Traceback (most recent call last):
  File "/home/nucleus/_dev/OpcUaCameraSlowControl/python-opcua/opcua/server/binary_server_asyncio.py", line 85, in _process_data
    ret = self.processor.process(hdr, buf)
  File "/home/nucleus/_dev/OpcUaCameraSlowControl/python-opcua/opcua/server/uaprocessor.py", line 72, in process
    msg = self._connection.receive_from_header_and_body(header, body)
  File "/home/nucleus/_dev/OpcUaCameraSlowControl/python-opcua/opcua/ua/uaprotocol_hand.py", line 631, in receive_from_header_and_body
    return self._receive(chunk)
  File "/home/nucleus/_dev/OpcUaCameraSlowControl/python-opcua/opcua/ua/uaprotocol_hand.py", line 662, in _receive
    self._check_incoming_chunk(msg)
  File "/home/nucleus/_dev/OpcUaCameraSlowControl/python-opcua/opcua/ua/uaprotocol_hand.py", line 583, in _check_incoming_chunk
    self.channel.SecurityToken.ChannelId))
UaError: Wrong channel id 6, expected 7
('Cleanup client connection: ', ('127.0.0.1', 43877))
```
And client is disconnected.

oroulet, 2016-06-15:
No this is the other one. Where the channel is wrong not the token. Can you do a new recording ? And make sure the error is about channel? I will have a look at the channel stuff now

nuccy, 2016-06-16:
Ok, testing again with capturing of the traffic.

oroulet, 2016-06-16:
I forgot to initialize a variable!!!!!. Sorry. Can you update and try again?
I should find a way to add a test for this...

nuccy, 2016-06-16:
Ok, restarted.

nuccy, 2016-06-16:
Two hours had passed, no disconnects occurred, no errors popped up so far.

nuccy, 2016-06-16:
oroulet,
it seems now it works fine, the opcua server under the same load was successfully serving data to client for around 6 hours, I guess the issue can be now closed.

Thank you.

oroulet, 2016-06-17:
You mean even the high load issue seems gone?

nuccy, 2016-06-17:
I guess yes, at least before after around an hour the client was disconnected and errors were thrown by the python opcua server, now it was connected the whole day (6 hours actually) I didn't got any errors from both sided.

closed 2016-07-26 by oroulet