FreeOpcUa/python-opcua #202: Timestamp in datachange notification

Johen8, 2016-06-06:
Hey, I just noticed that in the Subscription class we trigger the `datachange_notification` like this:
```python
if hasattr(self._handler, "datachange_notification"):
    event_data = DataChangeNotif(data, item)
    try:
        self._handler.datachange_notification(data.node, item.Value.Value.Value, event_data)
```
But I saw that `item.Value` has some useful information that we could pass to the notification, such as the Timestamp. The normal event notifications contain this information, is there a reason that we don't pass it when notifying for data changes?

oroulet, 2016-06-06:
The value itself is available as it's own argument but everything should be
available in event data structure. Let us know if something is missing

Johen8, 2016-06-06:
No, I don't think anything is missing, I just think that instead of
```python
self._handler.datachange_notification(data.node, item.Value.Value.Value, event_data)
```
We should do
```python
self._handler.datachange_notification(data.node, item.Value, event_data)
```
So that whoever is using the `datachange_notification` from the Subscription has some more information to work with.

oroulet, 2016-06-06:
It was an early choice to not expose variants and data value objects in the
high level python api but still have it available in one way or another. If
we change this we should do it everywhere. But I am so far quite happy with
the api as-is

Johen8, 2016-06-07:
Yes, I understand what you are saying. My main issue was with the timestamps. Say we have a queuesize of 10, when we receive the data changes for a subscription, we will get 10 values that we have no idea when in time they happened, or even if they happened in order (is there a guarantee that we get them in order at least?), but yes, I understand that changing this now would be a lot of unnecessary work.

closed 2016-06-12 by oroulet

cedricleroy, 2016-07-29:
I am not sure to do the correct things, but I get no timestamps doing the following: 

```python
def datachange_notification(self, node, val, data):
    print data.monitored_item.Value.SourceTimestamp
    print data.monitored_item.Value.ServerTimestamp
    print data.monitored_item.Value.Value
    print data.monitored_item.Value.StatusCode
```