FreeOpcUa/python-opcua #208: question about opc ua

politico6, 2016-06-14:
Hello! I have a question, is essential to load the address space from xml ? Is this essential for the standard OPCUA ??

eg: https://opcfoundation.org/UA/schemas/1.02/Opc.Ua.NodeSet2.xml

Can I load the address space from sqlite for example ? or other data source – database ???

Thanks!

politico6, 2016-06-14:
My question is: if I continuosly save the value of a digital signal (coming for example from a Arduino or Raspeberry) in a database, for example SQLITE, and when I start my OPCUA server, I take from that database this variable and its value (creating an UA object, obviously)

Do I meet the standads of OPC foundation??

Thanks

oroulet, 2016-06-15:
Opcua is a communication standard. It is not intended r to use to save your
data. I would make the core of my software independent of opcua, use db,
whatever and populate ua address space programmatically at startup

destogl, 2016-06-15:
Here is nobody from OPC foundation (as far as I know), so answers here would be interpretation of standard.

In my point of view I think there is no problem, because how server manages data internal is not important at all to be according to standard (as @oroulet already said)

zerox1212, 2016-06-15:
There is a method to dump your address space to a binary file. I spent a lot time making it work for this situation. If you are just testing something it will be much easier than making an entire storage implementation.

When you software stops, dump the address space. When your software starts, use a try block to load the address space again.

Here is example code from my project.

Before calling ```server.start()```
```python
        # populating our address space
        try:
            self.server.load_aspace("address_space.bin")
            print("loaded the address space backup")
        except:
            print("No address space backup found, importing XML instead")
            # import nodes from xml
            self.server.import_xml("config_OPCUA_v6.xml")```

Shutdown:
```python
    def shutdown(self):
        print("Shutting Down OPC UA")
        driver_ModbusRTU.poll_enable = False
        self.scan_off()
        self.server.stop()
        self.server.dump_aspace("address_space.bin")
        sys.exit(1)```

politico6, 2016-06-15:
thanks to all!!

destogl, 2016-07-06:
Seams solved.

closed 2016-07-06 by destogl