FreeOpcUa/python-opcua #209: Create custom objects

bitkeeper, 2016-06-15:
The Node.add_object let you only create object of the type BaseObjectType. It would be nice if you also could create other types by providing an objecttype.

Like:
  myobj = objects.add_object(idx, "MyCustomObject", myobjectype_nodeid)

Example and patch are included.

[customobjecttype_example.zip](https://github.com/FreeOpcUa/python-opcua/files/315627/customobjecttype_example.zip)

[manage_nodes_customobjecttypes.py.zip](https://github.com/FreeOpcUa/python-opcua/files/315593/manage_nodes_customobjecttypes.py.zip)

destogl, 2016-06-15:
Could you create Pull request?

oroulet, 2016-06-15:
A few months ago I started a method called instantiate  in instantiate.py ( not finished ). is it what you are looking it? Does your code replace it? Or should we merge these things?

bitkeeper, 2016-06-15:
@destogl : pull request is done

@oroulet : differs from instantiate.py. The only thing this patch do is adding an additional argument to the add_object for providing a NodeId for the TypeDefinition.

closed 2016-07-01 by bitkeeper