FreeOpcUa/python-opcua #211: etoken does not support python2

hunterjackson, 2016-06-15:
In file client.py, under the class `Client`, in the function `activate_session()`. When the etoken is generated it uses the `bytes` function in a way that is only compatible with python3. 

To solve the problem I replaced this line  
`etoken = ua.pack_bytes(bytes(password, "utf8") + self._server_nonce)`

with this  
```python
try:  
    etoken = ua.pack_bytes(bytes(password, "utf8") + self._server_nonce)  
except TypeError:  
    etoken = ua.pack_bytes(bytes(password) + self._server_nonce)
```

I'd submit a pull request but I'm not sure if this is an all inclusive fix, there maybe some stuff with unicode that will be needed to handle all situations

oroulet, 2016-06-15:
Should be possible to find a syntax that works both for python 2 and 3...
password.encode('utf-8').. maybe?

hunterjackson, 2016-06-15:
good thought! that works great for python2 I  just checked

hunterjackson, 2016-06-16:
Hi @oroulet I did a search on the project and I'm pretty sure I found one more instance of the bytes function only being used in a python3 compatible way.

client.py line 428  
`passwd = bytes(password, "utf8")`

Also, and I don't mean to bug you, but do you have a release date set in mind for this bug fix?

oroulet, 2016-06-16:
Can you make a pull request? I have to merge the pull requests, then do a new release. We just fixed an important bug in client, and improved event stuff

hunterjackson, 2016-06-16:
done https://github.com/FreeOpcUa/python-opcua/pull/216

closed 2016-06-20 by hunterjackson