FreeOpcUa/python-opcua #212: question about certificate

politico6, 2016-06-15:
Hi, I’m creating a opcua server for arduino, that will gladly share with all the community when it will be ready. I have two doubts/questions about the certificates: 

1) I have used example certificates (.der) and (.pem) given in the repositories. When I access the server through a client (I’ve used uaexpert client) it connects but the following error occurs: 'bad certificate hostname invalid”. My question is: this error occurs because my certificate is an auto-signed certificate? This is, it is not verified by any authority (for example verisign) 

2) I’ve tried to create other certificates but when I execute the server, an error occurs because it hasn’t been created properly. Could someone guide me in the process of creating it? Could someone guide me about how these certificates work? 

Thank you in advance to everyone

destogl, 2016-06-16:
I can answer you on 1). This should not be a problem with UA Expert since it has possibility to accept self-signed certificates. Problem is in hostname. Do you have spme hostname specified for your Arduino server?

politico6, 2016-06-16:
first gives the error 'bad certificate hostname invalid' ( opcions: cancel or ignore), and then the client connects without problem. I don't spme hostname specified .... maybe is the problem.

politico6, 2016-06-16:
solved:
 " the error refers to the SubjectAlternativeName extension, which shall contain the server's hostname(s) and/or IP addresses. If you connect to the server using it's IP address and the certificate only contains the hostname (or the other way round), this error will be thrown"

http://forum.unified-automation.com/topic1652.html

closed 2016-07-06 by destogl