FreeOpcUa/python-opcua #218: BadNodeIdInvalid

davidschranz, 2016-06-20:
Siemens supplies with its CNC an OPC UA Server called AccessMyMachine. I was able to readout values using freeopcua-0.10.11 without any problems on one machine. On another machine I get an error BadNodeIdInvalid on some values. Using a test client from UA called UAExpert I can readout all values from both machines.

node=client.get_node( "ns=2;s=/DriveVsa/DC/R0035[u5]")
value=node.get_value()
opcua.common.uaerrors.UaStatusCodeError: The syntax of the node id is not valid.(BadNodeIdInvalid)

Could anyone give me a hint how to get this resolved.

zerox1212, 2016-06-20:
When you subscribe to the node in UAExpert what do the NodeId attributes look like?

For example here is a screenshot of a OPC UA test server. This should give you a clue on what to pass to the ```get_node(0)``` method.

davidschranz, 2016-06-21:
Here the screenshot of the attributes:
[image]

oroulet, 2016-06-21:
in your screenshot the identifier is /DriveVsa/DC/R0035 not
/DriveVsa/DC/R0035[u5].
The error seems to say that your server does not support the "["
character....

davidschranz, 2016-06-21:
This server somehow only shows the first parameter of an array. The other values would have to be with brackets:
[image]

I have two servers, whereas the python call works on one server but not on the other one. Using the UAExpert, I get a result back from both servers:
[image]

oroulet, 2016-06-21:
OK. I think next step is to install wireshark and record 2 sessions, one with uaexpert and one with python-opcua while reading the same node. Then attach the result to the bug. Then we will see if we send something different than uaexpert

zerox1212, 2016-06-21:
Now that I see your server structure, why not call the ReadVar method with python instead of ```get_value()``` ?

davidschranz, 2016-06-22:
I will have access to the failing machine next Friday. Attached are already the Wireshark recordings including the test scripts of the machine that works. Thank you both for your support.

[Server1 Results.zip](https://github.com/FreeOpcUa/python-opcua/files/327238/Server1.Results.zip)

oroulet, 2016-06-22:
We need a record of a get_value from uaexpert too on the machine that does not worl. Simply click refresh on the attribute view.

davidschranz, 2016-06-27:
In short, the problem of the failing machine was on the server side.
UAExpert also failed on the get_value when added as a custom node. The last time I only tested the call_method with it. The workaround to use the call_method on the python side worked.
Thank you both @oroulet and @zerox1212 to get my problem resolved!

destogl, 2016-07-06:
After last comment seams solved.

closed 2016-07-06 by destogl