FreeOpcUa/python-opcua #227: ServerDiagnostics question

TheAtlantis, 2016-07-01:
Hi, 

I have set Server/Auditing and Server/ServerDiagnostics/EnabledFlag to 1 and with an external OPC UA Client (Unified Automation UaExpert) I try to subscribe to Server/ServerDiagnostics/CurrentSessionCout for exemple. 
I gets no diagnostic value ( same for others nodes ). 

Do you have an idea to get diagnostics ?

Michel

oroulet, 2016-07-01:
The server has absolutely no code for server diagnostic. If you know how it
should be used, patch are welcome

TheAtlantis, 2016-07-01:
Ok, thanks.

zerox1212, 2016-07-01:
Somewhat related, how can we properly implement ServerCapabilities? We should make an effort here to populate this object with the capabilities of the python OPC UA server.