FreeOpcUa/python-opcua #228: Custom enums

bitkeeper, 2016-07-01:
Someone  experience with how to implement a custom enumeration ?
Especially in such way that available enum values can parsed by the client.

oroulet, 2016-07-01:
Interesting question. I have never tried but you should be able to create custom node subtype of enum type and then reuse this in variable. But variable take a variant type. Which is not a node ID but can be a VarianTypeCustom ... Let us know what you find out

oroulet, 2016-07-01:
And a variant type id must be <63 because of encoding so there is something I have not completely understood here

oroulet, 2016-07-02:
I had a look and it looks like the spec i bit broken here and everything is left to the server.
Make a subtype of enum. add its property (EnumValues or EnumString) to define possible values. Then when you create a variable use that as DataType and a compatible VariantType. It looks like it is then the servers job to accept or not values which are part of enum or not...
We could add a create_enum() function, but the spec seems so broken there, there is probably not point doing it...

bitkeeper, 2016-07-04:
What I trying to reach is that a client (in my case a Scada application) like UAExpert can use the enum values for a picklist. 
For example below is a custom enumeration from UACPPserver (demo server):
[image]

bitkeeper, 2016-07-04:
I used you suggestion as starting point, I tried two directions:

1. With UaModel create a address space with a enum plus instance of the enum as vairable and import it in python-opcua server.
2. Hardcoded the enum and a variable of it in python-opcua server.

bitkeeper, 2016-07-04:
The schema import resulted in empty string list in EnumStrings in /Types/DataTypes/BaseDataType/Enumeration/MyEnum/EnumStrings. Probably something to do with the import. It looks like nothing is done with the definition node ?

Fragment of the used address schema:
```xml
  <UADataType NodeId="ns=1;i=3002" BrowseName="1:MyEnum">
        <DisplayName>MyEnum</DisplayName>
        <Description>Demonstrates enums</Description>
        <References>
            <Reference ReferenceType="HasProperty">ns=1;i=6001</Reference>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=29</Reference>
        </References>
        <Definition Name="1:MyEnum">
            <Field Name="ok" Value="0"/>
            <Field Name="run" Value="1"/>
            <Field Name="error" Value="2"/>
        </Definition>
    </UADataType>
    <UAVariable DataType="LocalizedText" ParentNodeId="ns=1;i=3002" ValueRank="1" NodeId="ns=1;i=6001" ArrayDimensions="3" BrowseName="EnumStrings">
        <DisplayName>EnumStrings</DisplayName>
        <References>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
            <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=3002</Reference>
        </References>
        <Value>
            <uax:ListOfLocalizedText>
                <uax:LocalizedText>
                    <uax:Text>ok</uax:Text>
                </uax:LocalizedText>
                <uax:LocalizedText>
                    <uax:Text>run</uax:Text>
                </uax:LocalizedText>
                <uax:LocalizedText>
                    <uax:Text>error</uax:Text>
                </uax:LocalizedText>
            </uax:ListOfLocalizedText>
        </Value>
    </UAVariable>
```

bitkeeper, 2016-07-04:
The result of the coded enum: the short summary; everyhting looks fine in UaExpert, only no picklist when trying to change the values..

I created the example code below:

**Create a custom Enumeration datatype**
```python
    # --------------------------------------------------------
    # create enum data type
    # --------------------------------------------------------
    enums = server.get_root_node().get_child(["0:Types", "0:DataTypes", "0:BaseDataType", "0:Enumeration"])

    # 1. Create Enum Type
    parent = enums
    nodeid = ua.NodeId.from_string("ns=1;i=26")
    qname = ua.QualifiedName.from_string("1:MyEnum2")
    objectype = ua.NodeClass.DataType
    # Node.node.add_subtype couldn't be used, because Node.NodeClass can't be set to ua.NodeClass.DataType
    # Create custom version of manage_nodes._create_object to get this example working
    myenum_type = node.Node(parent.server, m_create_object(parent.server, parent.nodeid, nodeid, qname, objectype))

    # 2. Create EnumStrings
    es = myenum_type.add_variable(idx, "EnumStrings" , [ua.LocalizedText("ok"), ua.LocalizedText("idle")])
```
Issue: The `Node.add_subtype` sets the `NodeClass `to  `ua.NodeClass.ObjectType`. Don't know how to modify to `ua.NodeClass.DataType` after creation. Create temp workarround by create a custom `_create_object`.

**Create an object with an instance of the custom Enumeration datatype**
```python
    # --------------------------------------------------------
    # create object with enum variable
    # --------------------------------------------------------
    # 1. creat object
    myobj = objects.add_object(idx, 'Base2')

    # 1. creat enum variable
    parent = myobj
    nodeid = ua.NodeId.from_string("ns=1;i=6005")
    qname = ua.QualifiedName.from_string("1:MyEnum2Var")

    datatype = myenum_type.nodeid
    val = 0

    # Node.add_variable doesn't allow setting the DataType setting by hand.
    # Create custom version of manage_nodes._create_variable to get this example working
    myenum_var = node.Node(parent.server, m_create_variable(parent.server, parent.nodeid, nodeid, qname, val, datatype, isproperty=False))
    myenum_var.set_writable()
    myenum_var.set_value(1)
```
Issue: The `Node.add_variable ` doesn't allow setting the DataType directly. creation. Create temp create a custom `_create_variable`. Don't know of using a custom variant solves this.

Maybe it has also something to do with definition part of a node. Have to investigate how that part works, in standard as in implementation.

oroulet, 2016-07-04:
Both should work in theory. If it does not, try to find differences between your custom enum and the one above. This is probably a small thing. 

But I am not sure your example above is a custom enum. It looks like the standard server state enum defined in standard address space. Uaexpert might have a hard coded solution for it....

oroulet, 2016-07-04:
Add_xxx method are high level and you may need to call ua method directly. Look at code and make your own function

bitkeeper, 2016-07-04:
@oroulet can you explain a little bit to me ?

The fragment blow show a typical use of the definition:
```xml
       <Definition Name="1:MyEnum">
            <Field Name="ok" Value="0"/>
            <Field Name="run" Value="1"/>
            <Field Name="error" Value="8"/>
        </Definition>
```
It contains the actual values that should be for the value of the enumeration.
Where the EnumStrings is just a list of string translations, linked to the index if the `definition/field` or `definition/field[@name] `.

In xmlparser.py `NodeData.__init__`  there is a place holder for Definition tags: `self.definition = []`. The definition tags are indeed parsed into this placeholder. But after that it isn't used. Is this not finished or not required ?

oroulet, 2016-07-04:
> Is this not finished or not required ?

Looks like you found something not finished. From what I see from your screenshot we should create a child property with correct making, type and value... but I haven't looked much into it...

destogl, 2016-07-04:
The function ``_create_object`` should be modified to enable setting the ``NodeClass`` to ``ua.NodeClass.DataType``. But looking at it, it starts to be complicated therefore would be better to add new method called ``_create_type`` where new types will be added, so that also ``add_subtype`` uses this new method.

oroulet, 2016-07-04:
@destogl yes! Create type or create data type

destogl, 2016-07-04:
@oroulet This could be the same function because the difference is very small (I expect!) and with 1-2 "if-s" per function we can live :)

oroulet, 2016-07-04:
@destogl I let you have a look ;-)

bitkeeper, 2016-07-04:
@oroulet Ok the missing Definition isn't problem ([AddressSpaceInterchangeXML](http://www.commsvr.com/InternetDSL/commserver/P_DowloadCenter/P_Publications/P-150101E-AddressSpaceInterchangeXML.pdf) explains about the Definition tags).

As always the devil is in the details. The qname of EnumStrings was wrong. After changing it to "0:EnumStrings" custom enumerations works like a charm. At least for the coded version.
```python
    # 2. Create EnumStrings
    es = myenum_type.add_variable("ns=2;i=2005", "0:EnumStrings" , [ua.LocalizedText("ok"), ua.LocalizedText("idle")]
```
UaExpert show the possible value list with text descriptions.

This is still without the custom variants type. For the client maybe I will try to generate a IntEnum on the fly. First I if have to integrate the workarrounds. To prevent double work @destogl are you providing the NodeClass patch ?

Thanks for the support !

bitkeeper, 2016-07-04:
And the proof of course ;-):
[image]

destogl, 2016-07-05:
Check #231 I added support for custom data types. There are not tested yet, because I didn't have time for it.

You can create custom types using:
```
custom_dtype = base_dtype.add_subtype(2, 'MyFooDataType', ua.NodeClass.BaseDataType)
```

Now when I think, it is possible to decide between NodeClasses based on parent... but for not I think is OK.

destogl, 2016-07-05:
@bitkeeper could you create example for custom enums? This would be great!

bitkeeper, 2016-07-05:
@destogl no problem to create an example, only it will require an additional fix. `Node.add_variable` is missing support for providing the datatype. The datatype is now based on the provided value in `manage_node._create_variable` :
```python
  attrs.DataType = _guess_uatype(val)
```

bitkeeper, 2016-07-05:
@oroulet found the reason why the enums with xml import doesn't work. The value parser doens't support values that contain a ListOfLocalizedText.

destogl, 2016-07-05:
How are you providing arguments? As python values or Variants?

bitkeeper, 2016-07-05:
@destogl as python values. 
If would be nice if the `Node.add_variable `supports the also an optional argument DataType.

`manage_nodes._create_variable` would then require an if:
```python
    if datatype:
        attrs.DataType = datatype
    else:
        attrs.DataType = _guess_uatype(val)
```

bitkeeper, 2016-07-05:
In case of the Enum the value is just a regular Int32, but the datatype is the nodeid of the used Enum type (see example earlier in this thread). 

As @oroulet indicated you could use VariantTypeCustom, but it is not always handy to be forced to use a VariantTypeCustom.

node.add_property(2, MyPropertyName, ua.Variant(0, ua.VariantType.Int32, nodeid_of_myenum)

bitkeeper, 2016-07-05:
@destogl to demonstrate the enum vs datatype, below a matching test:

```python
    def test_add_variable_with_datatype(self):
        o = self.opc.get_objects_node()
        v1 = o.add_variable(3, 'VariableEnumType1', ua.ApplicationType.ClientAndServer, ua.ObjectIds.ApplicationType)
        tp1 = v1.get_data_type()
        self.assertEqual( ua.NodeId(ua.ObjectIds.ApplicationType), tp1)
        
        v2 = o.add_variable(3, 'VariableEnumType2', ua.ApplicationType.ClientAndServer, ua.NodeId(ua.ObjectIds.ApplicationType) )
        tp2 = v2.get_data_type()
        self.assertEqual( ua.NodeId(ua.ObjectIds.ApplicationType), tp2)
```

destogl, 2016-07-05:
Please check also #233 for automatic method of custom types with or without properties. Please comment if I missed something.

destogl, 2016-07-05:
OK, but what is here datatype, can you create it? I guess `ua.ApplicationType.ClientAndServer` is value and `ua.ObjectIds.ApplicationType` is type?

destogl, 2016-07-05:
I am not sure for proposed extension with datatype, because the number of paramters will just rise. Still you will have problems with value than if you use datatype. this has IMHO a bit larger consequences than we thought.

destogl, 2016-07-05:
> Concerning types My original plan was to add methods 'add_xxx_type' for ex: 'add_object_type' 

My also, but than I say it is not so complicated and reduced a lot of duplicated code. I can change it if you think is better....

bitkeeper, 2016-07-05:
Hereby a full example for creating custom Enums:

``` python
'''
  This example demonstrates the use of custom enums by:
  - Create a custom enum type
  - Create an object that contains a variable of this type
'''
try:
    from IPython import embed
except ImportError:
    import code

    def embed():
        vars = globals()
        vars.update(locals())
        shell = code.InteractiveConsole(vars)
        shell.interact()

interactive = True

from opcua import ua, Server
from opcua.common import node
from enum import IntEnum

# Not required just for convenience
# Because this example is based on EnumStrings, the values should start at 0 and no gaps are allowed.
class MyEnum(IntEnum):
    ok = 0
    idle = 1

# helper method to automatically create string list
def enum_to_stringlist(a_enum):
    items = []
    for value in a_enum:
        items.append(ua.LocalizedText(value.name))
    return items

if __name__ == "__main__":
    # setup our server
    server = Server()
    server.set_endpoint("opc.tcp://0.0.0.0:4840/freeopcua/server/")

    # setup our own namespace, not really necessary but should as spec
    uri = "http://examples.freeopcua.github.io"
    nsidx = server.register_namespace(uri)

    # --------------------------------------------------------
    # create custom enum data type
    # --------------------------------------------------------
    enums = server.get_root_node().get_child(["0:Types", "0:DataTypes", "0:BaseDataType", "0:Enumeration"])

    # 1.
    # Create Enum Type
    myenum_type = enums.add_subtype(nsidx, 'MyEnum', ua.NodeClass.DataType)

    # 2.
    # Add enumerations as EnumStrings (Not yet tested with EnumValues)
    # Essential to use namespace 0 for EnumStrings !

    # By hand
    #     es = myenum_type.add_variable(0, "EnumStrings" , [ua.LocalizedText("ok"),
    #                                                       ua.LocalizedText("idle")])

    # Or convert the existing IntEnum MyEnum
    es = myenum_type.add_variable(0, "EnumStrings" , enum_to_stringlist(MyEnum))

    es.set_value_rank(1)
    es.set_array_dimensions([0])

    # --------------------------------------------------------
    # create object with enum variable
    # --------------------------------------------------------
    # get Objects node, this is where we should put our custom stuff
    objects = server.get_objects_node()

    # create object
    myobj = objects.add_object(nsidx, 'MyObjectWithEnumVar')

    # add var with as type the custom enumeration
    myenum_var = myobj.add_variable(nsidx, 'MyEnum2Var', MyEnum.ok, myenum_type.nodeid)
    myenum_var.set_writable()
    myenum_var.set_value(MyEnum.idle)  # change value of enumeration

    server.start()
    try:
        if interactive:
            embed()
        else:
            while True:
                time.sleep(0.5)

    except IOError:
        pass
    finally:
        server.stop()
        print("done")
```

If you want to try it out; doesn't work on the current head it requires fixes for 
- add_subtype  - requires extra argument for NodeClass. see PR #231
- add_variable - requires extra argument for datatype. Fix is made, no PR available yet

bitkeeper, 2016-07-05:
PR #234 contains the example.

closed 2016-07-26 by oroulet

lortolan79, 2019-02-25:
Hi to all,

I tried to implement this example but in the UaExpert I cannot chhose from a dropdown menu...
The CustomDataType seems to be created with success but the Property added to the DeviceSet does not show the enumeration
[image]

oroulet, 2019-03-03:
@lortolan79 are you sure that uaexpert supports custom enums? just asking, I never tried...

lortolan79, 2019-03-04:
Hi,

yes it seems to be supported (see attached). The example is created automatically by the library.

L.
[image]