FreeOpcUa/python-opcua #229: [Help] Custom Object Types

zerox1212, 2016-07-01:
In my project I need a way to differentiate between two BaseObjectType objects.
Object type 1 = normal object which will match a python class
Object type 2 = special "hardware" object type that requires a special python class

What is the best way to do this? Use a custom object type?

If I want to instantiate the correct python class based on what is in the address space I need to be able to know some how. I know that it would be better if my python program made the address space from it's classes instead of the other way around, but I didn't do my application that way and I'm too far along to change that.

bitkeeper, 2016-07-04:
Using an custom object type is nice way to differentiate between object types.
For example we are using a Scada application that use the object type to assign specific visualization.

For finding your nodes using something like, tewak it for your needs (folders with organize vs components):
```
    def find_your_types(node, type_to_find):
        found_nodes = []
        for child in node.get_children():
            if child.nodeid == type_to_find:
                found_nodes.append(child)

            found_nodes.extend(find_your_types(child, type_to_find))

        return found_nodes
   found_nodes = find_your_types(myroot, nodeid)
```

If get more complex when you are using something like:

- otype1 (based BaseObjectType )
- otype2 (based otype2 )
- otype3 (based otype1 )

And you want to find all otype1 based ones (that is including  otype2 and otyp3).

zerox1212, 2016-07-05:
Ok, that helps.

How do I import that from XML? I'm using XML importer to build my address space, but I don't know how to make my <UAObject> be of a different type. Furthermore how will the custom object type be added to the server? If you browse Python OPC UA Server Address types, only BaseObjectType is available.

bitkeeper, 2016-07-05:
See examples/customobject.xml how to define a new object type in your xml address space, or even better use UaModeler to define (Types/ObjecTypes/BaseObjectType -> Add New Type.

To make an instance of a specific type the UAObject should contain a reference to nodeid of the UAObjectType:
`   <Reference ReferenceType="HasTypeDefinition">ns=1;i=1002</Reference>`
Or you can also created the instance from UaModeler with Add Instance and use the Type Definition selector (choose Add another node),
After import the instances will automatically be created.

You can also mix it with UAObjectType in the XML and use that type definitions from python code.

closed 2016-07-16 by zerox1212