FreeOpcUa/python-opcua #230: [Server] Add Object and Object Type

maljac, 2016-07-02:
Hello, 

when I create a new object (`add_object`) I can now pass a custom object type as third argument (thanks to #210 and #215). But, when I add this object and the object type has for example one variable, the variable is not created under the newly created object.

Example:

```
...
obj_type = ua.NodeId.from_string('ns=0;i=2391')  # ProgramStateMachineType, which has some more children than my example above
myobj = objects.add_object(idx, "MyObject %s" % i, obj_type)
```

Instead of having several variables etc. under the new object `MyObject` I have none.

Please let me know if this is (a) currently not supported or (2) a bug.

oroulet, 2016-07-02:
This is currently not supported this is supposed to be part of the instanciate method in instanciate.py. But it is not finished, since I have not had used for it yet... feel free to finish it and send PR

closed 2016-07-26 by oroulet

maljac, 2016-07-26:
@oroulet : Why did you close the issue? Is the functionality implemented?

oroulet, 2016-07-26:
https://github.com/FreeOpcUa/python-opcua/pull/244

should work now. either using instantiate method or node.add_object

maljac, 2016-07-26:
Great! Just checked it.