FreeOpcUa/python-opcua #246: populating address space with an xml file and use its variables.

brochet, 2016-07-18:
Hello,

Thank you very much for your python-opcua client&server examples.
I'm a beginner in python and opc ua.
I created a server and I had generated an xml file to populate my address space (I use UaExpert Client).
[image]
I used in my python code: server.import_xml("my_file.xml") to import all nodes.
But now I want to use the nCommand variable in my python code, but I tried lots of different things to retrieve this variable in my code but I did not succeed.
For example, when I change the value of this variable with uaExpert client, I want to retrieve this value in my code. 
a view of my variable in my xml file:
[image]

Thanks in advance for your help.

zerox1212, 2016-07-18:
Can you post your code which tries to get the value of nCommand?

brochet, 2016-07-19:
I read the xml file directly with minidom, but it's not the good way because I want to read the value when the UaExpert client  change this value.
[image]

oroulet, 2016-07-19:
read the server example file for example and you should be able to figure out.
if you know the path:
objects = server.get_objects_node()
mydevice_var = objects.get_child(["0:controller", "0:state", xxxxxxx]) 
if you  know the node id:
server.get_node(mynodeid)

brochet, 2016-07-19:
I have an error. @oroulet it's possible to explain me how to fulfil get_child(["0:controller", "0:state", xxxxxxx]) in my case? 
[image]

oroulet, 2016-07-19:
your sample does not show the path but it shows the node id so just write
var = server.get_node("ns=2;i=9002")
then look at the var object

brochet, 2016-07-19:
Yes of course, but I don't know what to put in the place of : "0:controller", "0:state" and xxxxxxx to read the value equals to 10 in this case.

oroulet, 2016-07-19:
something like 
objects.get_child(["0:PNC1_sml", "0:DeviceState", "0:Main", "1:ctrl", "1:nCommand"])
0 is the namespace of the used browsename, seems to be 1 or 0 in you example but your screenshort does show it. it might be different for every node

oroulet, 2016-07-19:
and get value with var.get_value()

brochet, 2016-07-19:
Thank you very much !

closed 2016-07-19 by brochet