FreeOpcUa/python-opcua #247: BadResponseTooLarge wxecption while trying to connect to open62541 server

maljac, 2016-07-22:
Hi, 

I get the following exception when I try to connect to a open62541 opc-ua server:

```
Message MessageChunk(Header(type:b'MSG', chunk_type:b'A', body_size:20, channel:49), SequenceHeader(SequenceNumber:2, RequestId:2 ), SymmetricAlgorithmHeader(TokenId:53 ), 8 bytes) aborted: MessageAbort(error:StatusCode(BadResponseTooLarge), reason:None)
```
Client script:
```
from opcua.client.client import Client

c = Client('opc.tcp://127.0.0.1:16664')
c.connect()
```

Used server: open62541 example server, branch 0.2 (https://github.com/open62541/open62541.git)
Build and run with: 
```
cmake -D UA_BUILD_EXAMPLES=ON -D UA_LOGLEVEL=100 ..
make all
cd examples
./server
```
Log from the server:
```
[07/22/2016 09:31:03.791] info/network	Connection 4 | New connection over TCP from 172.17.0.1:43656
[07/22/2016 09:31:03.791] trace/network	Connection 4 | Process a HEL
[07/22/2016 09:31:03.791] trace/network	Connection 4 | Process a OPN
[07/22/2016 09:31:03.791] info/channel	Connection 4 | SecureChannel 3 | OpenSecureChannel: Opened SecureChannel
[07/22/2016 09:31:03.792] trace/network	Connection 4 | Process a MSG
[07/22/2016 09:31:03.792] trace/channel	Connection 4 | SecureChannel 3 | Final chunk message
[07/22/2016 09:31:03.792] debug/channel	Connection 4 | SecureChannel 3 | Session {12423BB8-3C36-BC69-A10A-30438E48B4EB} created
[07/22/2016 09:31:03.792] info/channel	Connection 4 | SecureChannel 3 | Could not send the message over the SecureChannel with error code 0x80b90000
```

I tried to to set the MaxResponseMessageSize in ua_client.UASocketClient._send_request to a maximum value but with no success: 
```
request.Parameters.MaxResponseMessageSize = 4294967295
```
Since the open62541 server works fine with uaExpert I assume some configuration problems (?) in freeopcua?

What is interessting, that we get the **same error** when trying to connect with the open62541 **client** with the freeopcua **server**.

Any suggestions...

oroulet, 2016-07-22:
Have you tried to set message size to a lower value? Can you send out when
using a freeopcua server and c client?

maljac, 2016-07-22:
@message size:
Yes, I tried 0, the max value (see above) as well as the value defined in
open62541 server (10....). The behaviour is tge same for all values.

I also have tried to connect to a freeopcua server using the open62541 C
client. Here I get the same error if I try to read a node.

oroulet, 2016-07-22:
Since it is the c client and server that are not happy and through an error I think it is much easier for them to tell us what they do not accept.

destogl, 2016-07-22:
check open62541/open62541#726

maljac, 2016-07-22:
Further investigation showed, that open62541 uses 
SendBufferSize = `65535` 
ReceiveBufferSize = `65535`
MaxMessageSize = `65535`
MaxChunkCount = 1

while freeopcua uses:

SendBufferSize = `65536` 
ReceiveBufferSize = `65536`
MaxMessageSize = `0`
MaxChunkCount = 0

I am able to connect to open62541 server using the FreeOPCUA client when I change in `uaprotocol_hand.py` before mentioned parameters to:
```
        self.ReceiveBufferSize = 65535
        self.SendBufferSize = 65535
        self.MaxMessageSize = 65535
        self.MaxChunkCount = 0
```

oroulet, 2016-07-22:
In many cases 0 is supposed to mean unlimited or max. But it looks like it confuses a lot of implementations.....I already had to remove it several places

jpfr, 2016-07-23:
Fixed in open62541 on branches master and 0.2.
I can connect with the python-freeopcua default settings now.

@maljac I'm surprised that it worked for you with
```
        self.ReceiveBufferSize = 65535
        self.SendBufferSize = 65535
        self.MaxMessageSize = 65535
        self.MaxChunkCount = 0
```
I thought the error on our side was introduced by MaxChunkCount = 0.

maljac, 2016-07-24:
@jpfr I tested again, and the problem was not the ChunkCount but the MessageSize. Thus, using the open62541 server before the fix and changing freeopcua settings to
```
        self.ReceiveBufferSize = 65535
        self.SendBufferSize = 65535
        self.MaxMessageSize = 65535
        self.MaxChunkCount = 0
```
worked for me.

With the fix, the freeopcua client works with open62541 server. I will test the other way arround too.

maljac, 2016-07-24:
@oroulet @destogl : The issue can be closed.

closed 2016-07-24 by maljac

oroulet, 2016-07-24:
Was this a bug in c stack? Connecting should work out of the box

maljac, 2016-07-24:
Bug was in open62541 due to  wrong handling of `0` for MaxMessageSize and  MaxChunkCount.