FreeOpcUa/python-opcua #248: Add repository to FreeOpcUa

zerox1212, 2016-07-25:
Can we get a repository for a Free OPC Ua Modeler?

I am working on building a simple XML address space generator like UaModeler, except it will be free :smile: .

oroulet, 2016-07-26:
@zerox1212 done. you should have received an invitation.

oroulet, 2016-07-26:
@zerox1212 you should be able to push directly to it and merge pr

zerox1212, 2016-07-26:
I have access to it now. Thanks.

If anyone is willing to help I would appreciate it it. I'm a self taught programmer and I've never tried to make something like this, so it will probably start out bad.

My initial commit will be mostly a script just for myself, but I figured maybe it will be useful to others.

oroulet, 2016-07-26:
User pyqt5 then i can help

zerox1212, 2016-07-26:
I'm more worried about how to create well designed Element Tree program.

oroulet, 2016-07-27:
I though a bit about this. You need a kind of internal model in you UI, the most obvious thing is to use a kind of XML model but maybe it is a good idea to start a opcua server internally. This has several advantages:
* maintain consistence of model (no dead link, all required attributes, etc..) by using opc protcol and everything already build in server
* you can use the existing API: node.add_variable(xxxx)
* you will anyway need many things from standard namespace like types, etc...
* add scripting possibilities to create nodes
* This splits you application in two quasi independent parts:
   * An UI to add nodes to internal OPC-UA server server (we can probably reuse code/widgets from existing client GUI)
  * a Method to export nodes to XML which might also be usefull in other scenarios

oroulet, 2016-07-27:
@zerox1212 maybe you can start on exporting nodes from server to XML so I can do the first start of UI by copying what I can from python GUI?

zerox1212, 2016-07-27:
I thought about doing it this way as well. Another reason to do it this way is that we can backup a server address space to XML instead of binary (pickle).

For now I will push my personal code to the master because it's already written. My small program takes a .CSV and makes it a simple address space in XML.

destogl, 2016-07-27:
> I thought about doing it this way as well. Another reason to do it this way is that we can backup a server address space to XML instead of binary (pickle).

It would be nice to look at XML communication if you are already converting the address space to XML. I think this could be done painlessly in parallel (if you like :) )

zerox1212, 2016-07-28:
>It would be nice to look at XML communication if you are already converting the address space to XML. I think this could be done painlessly in parallel (if you like :) )

I can try, but I would need significant help to get started on that because I'm not on the same level as the other developers here.

zerox1212, 2016-07-28:
Can you make a new python-opcua release to PYPI? Seems the current release is missing some Node methods that I need for creating XML.

For example ```node.get_user_access_level()``` is missing. For now I'll use the latest master in my project.

zerox1212, 2016-07-28:
Can move discussion to https://github.com/FreeOpcUa/opcua-modeler/issues/2 to keep things clean.

closed 2016-08-15 by oroulet