FreeOpcUa/python-opcua #250: [Server] Import XML and generate namespace

maljac, 2016-07-26:
Hi, 

I am currently working with several companion specifications and trying to load them via `server.import_xml(filename)`.
Since the companion standards use the namespaceId 1 in their spec, the xml import always tries to import exactly in the defined namespace index that is given in the file instead of in the namespace index which must be generated by the server on startup. 

As I read [this](https://opcfoundation.org/forum/opc-ua-companion-standards/namespaces-in-companion-specifications/#p197) it seems that the "correct" way is:

1. Load UA namespace (ns:0)
2. Create custom namespace (ns:1) -> `server.register_namespace(uri)` 
3. Load XML File and translate the nodes given in the xml file to correct namespace (ns:2)
4. Load XML File and translate the nodes given in the xml file to correct namespace (ns:3)

The order of the `Uri` elements in `<NameSpaceUris>` should determine the file local namespace indexes:

```
    <NamespaceUris>
        <Uri>http://somemodel.org</Uri>
        <Uri>http://anothermodel.org/</Uri>
    </NamespaceUris>
```

Then in the XML file, references to `ns=1;i=...` must be translated to the the index of the first uri, i.e. `http://somemodel.org` while nodes reference to `ns=2;i=...` must be translated to the index of the second uri.

The index of the uris must be managed by the server, i.e. the order of loading models determines the namespace index.

Since I am a novice in OPC, I hope I described the scenario correct. Just wondering how other handle this scenario.

oroulet, 2016-07-26:
I do not know what the correct behaviour is supposed to be but overridding the namespace specified in xml sounds broken. and increasing the default namespace at every import even more broken....

maybe we could offer option to override namespace index when importing, but what do we do if xml specifies several namespaces?

I am expecting that companion specification make XML files defining node TYPES  in namespace 0 (maybe 1??) and then users instanciate these new node types in the namespace they wish. But as I wrote earlier I have not read anything about this so if you find references saying otherwise let me know....

>>  i.e. the order of loading models determines the namespace index.

That soundsd definitely wrong

maljac, 2016-07-26:
I did a test with UAModeler and created two models:
```
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:s1="http://namespacetestone.org/namespacetest1/Types.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema">
    <NamespaceUris>
        <Uri>http://namespacetestone.org/namespacetest1/</Uri>
    </NamespaceUris>
    <Aliases>
        <Alias Alias="HasSubtype">i=45</Alias>
    </Aliases>
    <Extensions>
        <Extension>
            <ModelInfo Tool="UaModeler" Hash="NrIff1PEQBQVhu3oG1bXVA==" Version="1.4.4"/>
        </Extension>
    </Extensions>
    <UAObjectType NodeId="ns=1;i=1002" BrowseName="1:MyCustomObject">
        <DisplayName>MyCustomObject</DisplayName>
        <References>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=58</Reference>
        </References>
    </UAObjectType>
</UANodeSet>
```
```
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:s1="http://namespacetesttwo.org/namespacetest1/Types.xsd" xmlns:s2="http://namespacetestone.org/namespacetest1/Types.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema">
    <NamespaceUris>
        <Uri>http://namespacetesttwo.org/namespacetest1/</Uri>
        <Uri>http://namespacetestone.org/namespacetest1/</Uri>
    </NamespaceUris>
    <Aliases>
        <Alias Alias="HasSubtype">i=45</Alias>
    </Aliases>
    <Extensions>
        <Extension>
            <ModelInfo Tool="UaModeler" Hash="jgMyVmdZGMQzAOEXD9cHzA==" Version="1.4.4"/>
        </Extension>
    </Extensions>
    <UAObjectType NodeId="ns=1;i=1002" BrowseName="1:MyCustomSubObject">
        <DisplayName>MyCustomSubObject</DisplayName>
        <References>
            <Reference ReferenceType="HasSubtype" IsForward="false">ns=2;i=1002</Reference>
        </References>
    </UAObjectType>
</UANodeSet>
```
The second model imports the first and uses the object `MyCustomObject` to create a `MyCustomObject`.

The namespace id (!!) in the second file, matches the order of the imported namespace uri's. Thus, the reference to the `MyCustomObject` is defined as `ns=2;=1002`.

Please note: I do not meant that the namspace should be overriden, but the namespace index in the to be imported XML file must be replaced with the namespace id that is managed by the server for the given URI (the URI given in the `NameSpaceUris` element).

I have several released companion standards as well as some drafts: All define their model in namespace 1. But they seem to use the logic of the order of the `NamespaceUris` as described above.

So, actually it seems _quite_ simple to do this:

1. Load XML
2. Check if the defined NameSpaceUris are registered in the server
3. If they are not, create new namespace index and create the nodes given in the XML in the newly created index
4. If some URIs are registered, use the namespace index of the server that corresponds to the uri instead of the ns defined in the xml

One could also replace all `ns=...` in the loaded xml dom with the namespace uri and every time the server tries to create a new node one could check if the namespace is given as namespace index (int) or a namespace uri (valid uri string, or simply a string)

maljac, 2016-07-26:
Addition to my previous comment: A Model that uses 2 other models, exported by UAModeler

```
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:s1="http://yourorganisation.org/namespacetest3/Types.xsd" xmlns:s2="http://namespacetestone.org/namespacetest1/Types.xsd" xmlns:s3="http://opcfoundation.org/UA/DI/Types.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema">
    <NamespaceUris>
        <Uri>http://yourorganisation.org/namespacetest3/</Uri>
        <Uri>http://namespacetestone.org/namespacetest1/</Uri>
        <Uri>http://opcfoundation.org/UA/DI/</Uri>
    </NamespaceUris>
    <Aliases>
        <Alias Alias="HasSubtype">i=45</Alias>
    </Aliases>
    <Extensions>
        <Extension>
            <ModelInfo Tool="UaModeler" Hash="U876rPGC1QxHtqoA7tbyng==" Version="1.4.4"/>
        </Extension>
    </Extensions>
    <UAObjectType NodeId="ns=1;i=1003" BrowseName="1:MySubCustomObject">
        <DisplayName>MySubCustomObject</DisplayName>
        <References>
            <Reference ReferenceType="HasSubtype" IsForward="false">ns=2;i=1002</Reference>
        </References>
    </UAObjectType>
    <UAObjectType NodeId="ns=1;i=1002" BrowseName="1:AnotherOne">
        <DisplayName>AnotherOne</DisplayName>
        <References>
            <Reference ReferenceType="HasSubtype" IsForward="false">ns=3;i=1001</Reference>
        </References>
    </UAObjectType>
</UANodeSet>
```

oroulet, 2016-07-26:
> 
    Load XML
    Check if the defined NameSpaceUris are registered in the server
    If they are not, create new namespace index and create the nodes given in the XML in the newly created index
    If some URIs are registered, use the namespace index of the server that corresponds to the uri instead of the ns defined in the xml

sounds OK. pr welcome :-)
Importing namespaces is not implemented yet. but it is probably easy. 

> One could also replace all ns=... in the loaded xml dom with the namespace uri and every time the server tries to create a new node one could check if the namespace is given as namespace index (int) or a namespace uri (valid uri string, or simply a string)

as you want. I would like to avoid adding extra checks in address space, it is already slow enough. but adding checks in xml_importer is fine.

maljac, 2016-07-26:
Okay. I will try to find some time to propose a PR :-)

zerox1212, 2016-07-26:
Can someone give a quick explanation of the namespace design? I really don't understand it's purpose...

I get that having stuff like object TYPES in a separate namespace, but after that why do you need multiple namespaces?

oroulet, 2016-07-29:
@zerox1212 no idea either. it gives some more flexibility but it could have been avoided. I would have prefered to have namespaces (or simply groups) as node attributes instead of carrying them in all nodeids and browsenames

maljac, 2016-07-29:
@zerox1212 @oroulet : Imagine you have to model multiple complex objects (a pump, a cnc machine, etc.). You might want to reuse the model for specific domain. Thus you would create a information model for each domain:

1. Information model for pumps (http://mycompany.com/pumps/1.0
2. Information model for CNC machines (http://mycompany.com/cnc/1.0)

In order to distinguish both models the concept of namespaces is used. Thus, nodes can have the same id but only in different namespaces. As a unique identifier of these models a URI is used. 

Now imagine following scenario: You want to deliver a special type of pumps. But the information model for pumps does not allow modeling features for your special pump. Now you create a new information model, but you want to reuse all the stuff from the basic pump model (Inheritance). 

If you now want to import your special pump information model in the server, you must declare the dependencies of your model. Therefore you use the `NamespaceURIs` element. 

Since you can not assure a-priori to importing in your server which information model is saved in which namespace (exception is the UA standard namespace 0,), you must use the mentioned `NamespaceURIs` tag to describe the dependecies *relative* in your model.  Thus, the order of the URIs in the `NamespaceURIs` describes implicit how the server should handle the relations and depencies of the nodes in your model. The server resolves the namespace numbers in your nodes according to the URIs and the already known (=already imported) information models.

I am not sure if this (a) answers your question and (b) can be understood....

closed 2016-07-30 by maljac

PrudhviRajC, 2019-02-15:
Hallo @maljac , 

I am also facing the same issue, i.e., I have two namespaces in my XML file. If you dont mind could you please paste the python code for the importing the XML file and defining the namespaces?

Thank You

oroulet, 2019-02-15:
@PrudhviRajC this is supposed to be fixed now. namespaces in xml are imported and their index is updated depending on already existing namespaces in server

babalu27, 2020-08-06:
Hi, wondering if i could pick your brain about something. I am trying to emulate the way a PLC (Siemens) OPCUA server responds and i am loading the same xml model in both the PLC and the python api. 

In the xml model, the namespace for all the nodes is defined as ns=1. For some reason, all the PLC's i tried report the namespace as ns=4, even using a different API loads the model and serves out ns=4. How can i change the namespace in python to be ns=4 without modifying the XML model?

Thank you very much in advanced.

AndreasHeine, 2020-08-06:
thats normal!

babalu27, 2020-08-06:
> thats normal!
> 
> ns => is just a index of the namespace array if you import the models in a different order the ns changes too the ns-idx in a nodeset ist just inside this nodeset.

> thats normal!
> 
> ns => is just a index of the namespace array if you import the models in a different order the ns changes too the ns-idx in a nodeset ist just inside this nodeset.
> 
> just register dummy ns if you like but its ugly... typically you use the method "get namspace index" because the namespace you want has a special domain-name!
> 
> https://documentation.unified-automation.com/uasdkhp/1.0.0/html/nodeid_concept_2.png

That makes sense but in only have the one model to import into python server. I have now tried, 3 different vendors and they all import the same model and serve out ns=4.

Excuse my ignorance i am not a python guru, how do i go about creating the dummy namespaces? do i need dummy xml models? and uris? Thanks again.

AndreasHeine, 2020-08-06:
just register enough namespaces before your model 0=Std-Nodes 1=dummy1 2=dumm2 3=dummy3 4=your model

server.register_namespace("dummy1")
server.register_namespace("dummy2")
server.register_namespace("dummy3")
server.register_namespace("your model")

babalu27, 2020-08-06:
> just register enough namespaces before your model 0=Std-Nodes 1=dummy1 2=dumm2 3=dummy3 4=your model
> 
> server.register_namespace("dummy1")
> server.register_namespace("dummy2")
> server.register_namespace("dummy3")
> server.register_namespace("your model")
> 
> but its ugly and its not like opcua should be used...
> 
> if you like you can read more about "companion spec." in general this would clearify a lot for you!

Thank you very much! yes i agree it's not the cleanest but i think this will do it for now. I'm going to try it now.

zerox1212, 2020-08-06:
If you are trying to emulate what the PLC does you should probably make your python server mirror whatever is in NS 1,2,3 of the PLC.

babalu27, 2020-08-06:
> If you are trying to emulate what the PLC does you should probably make your python server mirror whatever is in NS 1,2,3 of the PLC. From what you are saying NS4 is the first namespace available for user nodes in the PLC.

Yes it seems like it. it always is ns=4.  I havent had any luck yet. no matter how many "dummy" namespaces i add i always end up with the model in ns=1.

AndreasHeine, 2020-08-06:
ok i guess i know what you want to do give me a sec to write it down ;)

AndreasHeine, 2020-08-06:
typically you have got a nodeset full of models, object types and structures as blueprints and then you instantiate from those blueprints to build up your own address space out of them (its like i said its like using a companion spec.)

babalu27, 2020-08-06:
oh man, that is definitely not what i have. i have an xml file that contains the data model with variables, properties, methods etc. it gets imported to PLC's off you go. The import part is working fine in py and i can browse the model correctly using UA expert so that part is perfect. All i need is the ns to  = 4 and donesky. should i consider changing the ns in the xml model?

AndreasHeine, 2020-08-06:
to be honest that sound really unusual... in opcua the ns-name (domain-name) is the leading part and not the ns-index!

so finally please listen:
if you import a xml with ns=4 inside your server will change that on import to the next free index in the ns-array so even if you are change ns in xml the server will overwrite it to avoid node-id collisions!!! thats why we in opcua work with companion spec. (with blueprints and instantiate them structured in namespaces) the only way to get it to ns=4 is to read the namespace-array and check that index 4 is not used, if it is used before import you will have no chance to get ns=4.... , if not you need to register namespaces till the next free ns-index is 4, then import your xml and it should have ns-index 4! i dont know your code i try to guess from the information you provide... if this was also not helpfull you should give us more information!?

babalu27, 2020-08-06:
Sorry i did not mean it was not helpful, completely on the contrary! i am very appreciative of the feedback and advice!. I think i'm close, the namespace array is ['http://opcfoundation.org/UA/', 'opc.tcp://localhost:61510/SampleServer', 'place_holder_ns_1', 'place_holder_ns_2'] i need to get the place holders in place before the sample server.

babalu27, 2020-08-06:
Got it! ended up being an order of operations stupid moment on my part. Thank you all!