FreeOpcUa/python-opcua #254: [Server] Import XML - Importing method arguments

maljac, 2016-07-27:
Hi, 

when importing a XML model via `server.import_xml(...)` which includes a UAVariable with `DataType=Argument` the following error is raised:
```
File "/home/jacobi/projects/opcua/python-opcua/opcua/common/xmlimporter.py", line 28, in import_xml
    self.add_variable(node)
  File "/home/jacobi/projects/opcua/python-opcua/opcua/common/xmlimporter.py", line 111, in add_variable
    attrs.Value = ua.Variant(obj.value, getattr(ua.VariantType, obj.valuetype))
  File "/usr/lib/python3.4/enum.py", line 255, in __getattr__
    raise AttributeError(name) from None
```
The error is introduced with commit 9398d46 which enables the import `ListOfExtensionObject`s. I can fix the import error by adding some values to the `uatypes.VariantType`  enum:

```
class VariantType(Enum):
    ...
    Argument = 296
```

Now, if I browse the node in UaExpert and try to read the InputArguments node of the method, I get following execption in the server:
```
Exception raised while parsing message from client, closing
Traceback (most recent call last):
  File "/home/jacobi/projects/opcua/python-opcua/opcua/server/binary_server_asyncio.py", line 85, in _process_data
    ret = self.processor.process(hdr, buf)
  File "/home/jacobi/projects/opcua/python-opcua/opcua/server/uaprocessor.py", line 82, in process
    return self.process_message(msg.SecurityHeader(), msg.SequenceHeader(), msg.body())
  File "/home/jacobi/projects/opcua/python-opcua/opcua/server/uaprocessor.py", line 103, in process_message
    return self._process_message(typeid, requesthdr, algohdr, seqhdr, body)
  File "/home/jacobi/projects/opcua/python-opcua/opcua/server/uaprocessor.py", line 175, in _process_message
    self.send_response(requesthdr.RequestHandle, algohdr, seqhdr, response)
  File "/home/jacobi/projects/opcua/python-opcua/opcua/server/uaprocessor.py", line 41, in send_response
    data = self._connection.message_to_binary(response.to_binary(), message_type=msgtype, request_id=seqhdr.RequestId, algohdr=algohdr)
  File "/home/jacobi/projects/opcua/python-opcua/opcua/ua/uaprotocol_auto.py", line 7868, in to_binary
    packet.append(fieldname.to_binary())
  File "/home/jacobi/projects/opcua/python-opcua/opcua/ua/uatypes.py", line 1178, in to_binary
    packet.append(self.Value.to_binary())
  File "/home/jacobi/projects/opcua/python-opcua/opcua/ua/uatypes.py", line 1020, in to_binary
    b.append(pack_uatype_array(self.VariantType.name, flatten(self.Value)))
  File "/home/jacobi/projects/opcua/python-opcua/opcua/ua/uatypes.py", line 101, in pack_uatype_array
    b.append(pack_uatype(uatype, val))
  File "/home/jacobi/projects/opcua/python-opcua/opcua/ua/uatypes.py", line 130, in pack_uatype
    return value.to_binary()
AttributeError: 'dict' object has no attribute 'to_binary'
```

Any ideas?

The xml file to reproduce the result:
```
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:s1="http://yourorganisation.org/extensionerror1/Types.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema">
    <NamespaceUris>
        <Uri>http://yourorganisation.org/extensionerror1/</Uri>
    </NamespaceUris>
    <Aliases>
        <Alias Alias="HasModellingRule">i=37</Alias>
        <Alias Alias="HasTypeDefinition">i=40</Alias>
        <Alias Alias="HasSubtype">i=45</Alias>
        <Alias Alias="HasProperty">i=46</Alias>
        <Alias Alias="HasComponent">i=47</Alias>
        <Alias Alias="Argument">i=296</Alias>
    </Aliases>
    <Extensions>
        <Extension>
            <ModelInfo Tool="UaModeler" Hash="HqP+z3HOAS5zZlvi3u3sHg==" Version="1.4.4"/>
        </Extension>
    </Extensions>
    <UAObjectType NodeId="ns=1;i=1002" BrowseName="1:MyObjectType">
        <DisplayName>MyObjectType</DisplayName>
        <References>
            <Reference ReferenceType="HasComponent">ns=1;i=7001</Reference>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=58</Reference>
        </References>
    </UAObjectType>
    <UAMethod ParentNodeId="ns=1;i=1002" NodeId="ns=1;i=7001" BrowseName="1:MyMethod">
        <DisplayName>MyMethod</DisplayName>
        <References>
            <Reference ReferenceType="HasProperty">ns=1;i=6001</Reference>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=1002</Reference>
            <Reference ReferenceType="HasProperty">ns=1;i=6002</Reference>
        </References>
    </UAMethod>
    <UAVariable DataType="Argument" ParentNodeId="ns=1;i=7001" ValueRank="1" NodeId="ns=1;i=6001" ArrayDimensions="1" BrowseName="InputArguments">
        <DisplayName>InputArguments</DisplayName>
        <References>
            <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=7001</Reference>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
        </References>
        <Value>
            <uax:ListOfExtensionObject>
                <uax:ExtensionObject>
                    <uax:TypeId>
                        <uax:Identifier>i=297</uax:Identifier>
                    </uax:TypeId>
                    <uax:Body>
                        <uax:Argument>
                            <uax:Name>Context</uax:Name>
                            <uax:DataType>
                                <uax:Identifier>i=12</uax:Identifier>
                            </uax:DataType>
                            <uax:ValueRank>-1</uax:ValueRank>
                            <uax:ArrayDimensions/>
                            <uax:Description/>
                        </uax:Argument>
                    </uax:Body>
                </uax:ExtensionObject>
            </uax:ListOfExtensionObject>
        </Value>
    </UAVariable>
    <UAVariable DataType="Argument" ParentNodeId="ns=1;i=7001" ValueRank="1" NodeId="ns=1;i=6002" ArrayDimensions="1" BrowseName="OutputArguments">
        <DisplayName>OutputArguments</DisplayName>
        <References>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=7001</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
        </References>
        <Value>
            <uax:ListOfExtensionObject>
                <uax:ExtensionObject>
                    <uax:TypeId>
                        <uax:Identifier>i=297</uax:Identifier>
                    </uax:TypeId>
                    <uax:Body>
                        <uax:Argument>
                            <uax:Name>Status</uax:Name>
                            <uax:DataType>
                                <uax:Identifier>i=6</uax:Identifier>
                            </uax:DataType>
                            <uax:ValueRank>-1</uax:ValueRank>
                            <uax:ArrayDimensions/>
                            <uax:Description/>
                        </uax:Argument>
                    </uax:Body>
                </uax:ExtensionObject>
            </uax:ListOfExtensionObject>
        </Value>
    </UAVariable>
</UANodeSet>
```

oroulet, 2016-07-27:
> I can fix the import error by adding some values to
theuatypes.VariantType enum:

Do not do that. That enum should only be used for variants, not for node
data types! There is something wrong here.. But I cannot have a look at
this before a few days

zerox1212, 2016-07-27:
Importing UA methods probably isn't tested. I'm not sure if it's even supported by the importer...

maljac, 2016-07-27:
xmlimporter.py :
```
elif node.nodetype == 'UAMethod':
                self.add_method(node)
```

Its not the method, the arguments für the method (in, out) that fail.

maljac, 2016-07-28:
If one of you guys could give me a hint where to look I would propose a PR.

zerox1212, 2016-07-28:
I tested your problem and it's obvious what is happening. The importer is trying to make a Variable that isn't a Variant which breaks the creation of the node.

Either the XML is wrong, or there is some special way to handle this that we must look for in the OPC UA spec.

maljac, 2016-07-28:
Since the UML is created by UAModeler and the comparison with companion
specs show same XML, we can assume that this is a special case. Anyway, I
will check the spec as you suggested.

zerox1212, 2016-07-28:
You are fast. You quoted my response before I edited lol. Please read again.

maljac, 2016-07-28:
Haha, okay. Thanks for the hint. I ll give it a try.

PR #257 (2016-07-29): [Server] Allow Methods to be imported via XML
Fixes #254
closed 2016-07-29 by oroulet