FreeOpcUa/python-opcua #258: [Server] Import XML - Dependency Order

maljac, 2016-07-29:
Hi, 

the current implementation of the XML Importer / Parser does not take into account the dependencies of the nodes defined in the xml. Currently it tries to create nodes according to the order of the XML elements.

Workaround: Import the files a second time.

Feature proposal: Respect the dependency tree defined in the XML nodes while importing.

oroulet, 2016-07-29:
I did not even know there was such a thing. How do you read the
dependencies?

zerox1212, 2016-07-29:
Can you post an example XML that would cause this?

maljac, 2016-07-30:
Example:
```
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:s1="http://myorg.de/simpleModel/Types.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema">
    <NamespaceUris>
        <Uri>http://myorg.de/simpleModel/</Uri>
    </NamespaceUris>
    <Aliases>
        <Alias Alias="Double">i=11</Alias>
        <Alias Alias="String">i=12</Alias>
        <Alias Alias="ByteString">i=15</Alias>
        <Alias Alias="LocalizedText">i=21</Alias>
        <Alias Alias="HasModellingRule">i=37</Alias>
        <Alias Alias="HasEncoding">i=38</Alias>
        <Alias Alias="HasDescription">i=39</Alias>
        <Alias Alias="HasTypeDefinition">i=40</Alias>
        <Alias Alias="GeneratesEvent">i=41</Alias>
        <Alias Alias="HasSubtype">i=45</Alias>
        <Alias Alias="HasProperty">i=46</Alias>
        <Alias Alias="HasComponent">i=47</Alias>
        <Alias Alias="MachineStateEnum">ns=1;i=3002</Alias>
        <Alias Alias="SpindleStatus">ns=1;i=3003</Alias>
        <Alias Alias="AxesPosition">ns=1;i=3004</Alias>
    </Aliases>
    <Extensions>
        <Extension>
            <ModelInfo Tool="UaModeler" Hash="c8VvmA89MWQ5CRq41HZn3w==" Version="1.4.4"/>
        </Extension>
    </Extensions>
    <UADataType NodeId="ns=1;i=3002" BrowseName="1:MachineStateEnum">
        <DisplayName>MachineStateEnum</DisplayName>
        <References>
            <Reference ReferenceType="HasProperty">ns=1;i=6003</Reference>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=29</Reference>
        </References>
        <Definition Name="1:MachineStateEnum">
            <Field Name="STOPPED" Value="0"/>
            <Field Name="RUNNING" Value="1"/>
        </Definition>
    </UADataType>
    <UAVariable DataType="LocalizedText" ParentNodeId="ns=1;i=3002" ValueRank="1" NodeId="ns=1;i=6003" ArrayDimensions="2" BrowseName="EnumStrings">
        <DisplayName>EnumStrings</DisplayName>
        <References>
            <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=3002</Reference>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
        </References>
        <Value>
            <uax:ListOfLocalizedText>
                <uax:LocalizedText>
                    <uax:Text>STOPPED</uax:Text>
                </uax:LocalizedText>
                <uax:LocalizedText>
                    <uax:Text>RUNNING</uax:Text>
                </uax:LocalizedText>
            </uax:ListOfLocalizedText>
        </Value>
    </UAVariable>
    <UADataType NodeId="ns=1;i=3003" BrowseName="1:SpindleStatus">
        <DisplayName>SpindleStatus</DisplayName>
        <References>
            <Reference ReferenceType="HasProperty">ns=1;i=6004</Reference>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=29</Reference>
        </References>
        <Definition Name="1:SpindleStatus">
            <Field Name="STOPPED" Value="0"/>
            <Field Name="RUNNING" Value="1"/>
        </Definition>
    </UADataType>
    <UAVariable DataType="LocalizedText" ParentNodeId="ns=1;i=3003" ValueRank="1" NodeId="ns=1;i=6004" ArrayDimensions="2" BrowseName="EnumStrings">
        <DisplayName>EnumStrings</DisplayName>
[... 234 lines cut ...]
        <DisplayName>SpindleState</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=2365</Reference>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=1002</Reference>
        </References>
    </UAVariable>
    <UAObject SymbolicName="DefaultBinary" NodeId="ns=1;i=5001" BrowseName="Default Binary">
        <DisplayName>Default Binary</DisplayName>
        <References>
            <Reference ReferenceType="HasDescription">ns=1;i=6010</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=76</Reference>
            <Reference ReferenceType="HasEncoding" IsForward="false">ns=1;i=3004</Reference>
        </References>
    </UAObject>
    <UAObject SymbolicName="DefaultXml" NodeId="ns=1;i=5002" BrowseName="Default XML">
        <DisplayName>Default XML</DisplayName>
        <References>
            <Reference ReferenceType="HasDescription">ns=1;i=6011</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=76</Reference>
            <Reference ReferenceType="HasEncoding" IsForward="false">ns=1;i=3004</Reference>
        </References>
    </UAObject>
</UANodeSet>
```

maljac, 2016-08-01:
Hi, 

it seems, that the import order should be depending on the nodes `ParentNodeId`. I read some docs related to references etc. and my interpretation of references is, that it can happen that a reference might target a node that does not / or not yet exist. Thus, my conclusion is that the `ParentNodeId` is the relevant information to build a dependency tree.

The example XML is a export from UaModeler. UaModeler does not always export the nodes in the correct order (i.e. the node with id `ns=1;i=6010` comes before it's parent `ns=1;i=6004`).

So, we could either (a) parse the xml and order the nodes according `ParentNodeId` or (b) try to import nodes and kind of _reschedule_ the import. I would vote for (a) if this is possible with the current XML import and parse process.

oroulet, 2016-08-01:
Is this an issue? I remember while implementing the importer that I would just create dangling references and hope that the target node would get created later.

zerox1212, 2016-08-01:
I'm not an expert with XML, but I think it would probably be easier to just check if the XML element has a ParentNodeId tag. If it does, check if that node exists, if it doesn't exist you would need to scan the XML for that specific node and create, then return to what you were doing. However seems like this could have poor performance.

maljac, 2016-08-02:
It turned out that its quite simple to implement. Because a node has 0 or 1 parentNodeId. 

I work on a PR, will submit this evening.

PR #267 (2016-08-05): [Server] Import XML - Respect depency of nodes (ParentNodeId)
Fixes #258
closed 2016-08-05 by oroulet