FreeOpcUa/python-opcua #265: Timestamps in Subscription hander

cedricleroy, 2016-07-31:
This may be the same than #202. Looks like timestamps cannot be retrieved in _datachange_notification_. I did try the following:
```python
def datachange_notification(self, node, val, data):
    print data.monitored_item.Value.SourceTimestamp
    print data.monitored_item.Value.ServerTimestamp
    print data.monitored_item.Value.Value
    print data.monitored_item.Value.StatusCode
```
And got None for _SourceTimestamp_ and _ServerTimestamp_, though I get timestamps when pulling data manually without going through the event handler.

oroulet, 2016-07-31:
Look at the MonitoredItem which is sent while subscribing. Maybe it does not request timestamps. If so, this is an error

cedricleroy, 2016-07-31:
In _create_monitored_items_:
```python
params.TimestampsToReturn = ua.TimestampsToReturn.Neither
```

It works properly using Both.

oroulet, 2016-07-31:
Can you send a PR?

maljac, 2016-08-02:
Can we close this?

closed 2016-08-03 by cedricleroy