FreeOpcUa/python-opcua #276: [Client/Question] Write on nodes that have a structured datatype

maljac, 2016-08-10:
I was wondering how to write on a node that has a structured type as datatype.

For example: The node `Server.ServerStatus.BuildInfo` is node with a structured datatype. How can I write on such nodes?

I found a the class `BuildInfo` in `uaprotocol_auto.py` which seems to represent the structure. Do I need a class for each structured object that I want to write / read?

oroulet, 2016-08-10:
Isn't a structured data type an extensionobject? ExtensionObjects can contain whatever, but the reader must know the structure to decode it correctly.

Look for ExtensionObjects _to_ binary and similar in code. It is used a few places. If you can come up with a higher level interface let me know

oroulet, 2016-08-10:
We could at least attempt to automate decoding. since the node id of the extension object should link to a node in namespace that contains the structure.

oroulet, 2016-08-17:
@maljac Have you found out something? Extension objects poins to a nodeid, so we can read the nodeid and its children define the content of the structure, but I have no idea how they should be ordered. Childs are not ordered.....

similar issue here https://github.com/FreeOpcUa/python-opcua/issues/287

oroulet, 2016-08-17:
BuildInfo for example has probably a type of BuildInfoType which is i=3052. This node has all the children with correct type. But the order of children is random, so how do we get the order....

oroulet, 2016-08-17:
@terrub @maljac OK I found out the order is the order of the node ids! i=3056  is before i=3057
so now it is quite simple to decode
1- read data type attribute, this is a node
2- read children of datatype node
3- order them by nodeid
4- for each node unpcack its type from binary data of extended object.
This should be integrated into binary_to_extensionobject, if the structure is not known we read address space and decode from it.
I let you have a look I have

maljac, 2016-08-17:
@oroulet I had no time looking at this issue. But is still important for my use-cases, thus I will try to solve this. I will check your suggested solution.

oroulet, 2016-08-17:
For encoding we probably need to make a function that generate a class from
a node type. Then extend Node.set_value to pack class into extension object
when type is extension object

oroulet, 2016-08-17:
Btw you can probably copy code from event structure generator. Make something generic.

oroulet, 2016-08-17:
Thinking about it the code should be placed inside variant class. If type is extension object we pack value in extension object.

SeetaiahB, 2017-07-12:
1. Can any one give an example to how to get the Body information from ExtensionObject? 
2. How to build input and output arguments for given method calling , if some body explain with example appreciated.

oroulet, 2017-07-12:
Look at examples. for exmaple this one: https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client_read-custom_structures.py

camheras, 2019-06-21:
Can we have an update for this issue because the example is not working for me