FreeOpcUa/python-opcua #279: ServiceFault after ActivateSessionRequest while using certificates

trevorhodges, 2016-08-11:
Hello,

Just started trying to work with the OPC UA client. Have been able to get the client working with my TOP Server OPC server with no authentication or certificates. Next step is to use certificates. When attempting to do so, I receive a ServiceFault response from TOP Server after sending the ActivateSessionRequest. All of this occurs when hitting the client.connect() line of code.

Below is the stack trace I'm receiving. Attached is a packet capture that I'm seeing. Note that client is 10.1.70.187 and the server is 10.1.70.159. The OPC server listening port is 49380
[ServiceFault.zip](https://github.com/FreeOpcUa/python-opcua/files/412395/ServiceFault.zip)

--------------------------------------------------------------------------
UaStatusCodeError                         Traceback (most recent call last)
<ipython-input-2-a9f21c90fee7> in <module>()
----> 1 client.connect()

/home/thodges/Programming/MQTT/scadamqttnode/opcuatests/opcuatests/lib/python3.5/site-packages/opcua/client/client.py in connect(self)
    215         self.open_secure_channel()
    216         self.create_session()
--> 217         self.activate_session(username=self.server_url.username, password=self.server_url.password, certificate=self.user_certificate)
    218 
    219     def disconnect(self):

/home/thodges/Programming/MQTT/scadamqttnode/opcuatests/opcuatests/lib/python3.5/site-packages/opcua/client/client.py in activate_session(self, username, password, certificate)
    387         else:
    388             self._add_user_auth(params, username, password)
--> 389         return self.uaclient.activate_session(params)
    390 
    391     def _add_anonymous_auth(self, params):

/home/thodges/Programming/MQTT/scadamqttnode/opcuatests/opcuatests/lib/python3.5/site-packages/opcua/client/ua_client.py in activate_session(self, parameters)
    244         request = ua.ActivateSessionRequest()
    245         request.Parameters = parameters
--> 246         data = self._uasocket.send_request(request)
    247         response = ua.ActivateSessionResponse.from_binary(data)
    248         self.logger.debug(response)

oroulet, 2016-08-11:
Does certificate login works with uaexpert?

cedricleroy, 2016-08-12:
I didn't test with uaexpert, but I have the same problem with TopServer (though getting a different error): 

_StatusCode(BadServiceUnsupported)_

Here is what I am getting from TopServer log:

_UaServer_EndpointCallback: SecureChannel 27972041 opened!
Security Mode 1 Security Policy http://opcfoundation.org/UA/SecurityPolicy#None_ 

What would be the next steps to nailing down this bug?

oroulet, 2016-08-12:
Use wireshark and record a working session with uaexpert and one not working from python. Then we should be able to spot the difference

oroulet, 2016-08-12:
Of course try to set up both clients in a similar way.