FreeOpcUa/python-opcua #280: Free OPC UA logo

zerox1212, 2016-08-11:
I made a logo/icon for Free OPC UA. What do you think?
[image]

I have it in various sizes. I was thinking of making a PR to https://github.com/FreeOpcUa/opcua-widgets, maybe a new folder called resources.

oroulet, 2016-08-11:
Great that you are looking at icons, they definitely need soon love.
My comment would be that I would like icons representing more what the software does. Rather UA or fUA than F.
Another possibility is to draw 2 aliens/machines exchanging data/communicating/arrows.
For the modeler would nice to represent a data model. Maybe a data tree?

oroulet, 2016-08-11:
Yes we should move all icons to a folder called resource.

zerox1212, 2016-08-11:
Ok. I will try have more descriptive icon. I try to keep a clean design, more of a mobile app style. Icons with to much text don't look nice. I will try to update it tonight and post it there.

oroulet, 2016-08-12:
this is fine for a logo. But I think I would have UA letters instead of F, this make it mych clearer for users what it is. Had anybody else opinion?

zerox1212, 2016-08-12:
Ok, thanks for the feedback.

oroulet, 2016-08-12:
Btw, have you any plan to work on xml stuff? I am eager to have a prototype

zerox1212, 2016-08-12:
I will work on it tonight and the weekend. Sorry for the delay. My job is controls engineer/PLC programmer and I had customers at our office the last week so I didn't have time for side projects.

zerox1212, 2016-08-15:
Is this good enough for now?
Default:
[image]

Client:
[image]

Modeler:
[image]

oroulet, 2016-08-15:
OK.  Maybe make a version with only UA too. and send the SVG fil

destogl, 2016-08-15:
This looks good!

zerox1212, 2016-08-30:
Do you need SVG? I used GIMP to create these so best I can give you is PNG. If you need SVG I can redraw it I guess.

zerox1212, 2016-09-17:
@oroulet can you let me know what you need? I drew these icons in a non vector format. If you require SVG please let me know so I can redraw them.