FreeOpcUa/python-opcua #281: Release 0.13.0

georgschoelly, 2016-08-11:
It seems still some way to version 1.0 (#103) and we've accumulated quite some changes sinec 0.12.0. Personally, I would be happy about another interim release. I've gone through the changes and found these changes, but I have to say I'm not qualified enough to understand all of the changes:

[client & server] 3b97ddc  add specific subclasses for all UaStatusCodeError
[server] 9c60ba3 add support for instantiating nodes with multiple parents
[server] bb5a2ff add support for instantiating DataTypes and Methods
[server] 7e9f42f, 14a33c9, f17f79e, b9b1be1 improved support for importing XML
and many more small bug fixes and improvements

Full list: https://github.com/FreeOpcUa/python-opcua/compare/cec49e0c36238ed02c2b8890b15f5f4da076f41a...master

oroulet, 2016-08-11:
> Full list: cec49e0...master

Good point. But maybe I should release a 1.0 just to emphasis that the
software is stable. I have used proprietary stacks more buggy at version 5
or 6

oroulet, 2016-08-12:
done, although I do not have a good overview of changes

closed 2016-08-12 by oroulet