FreeOpcUa/python-opcua #284: [help] OPC UA Object to Python Object

zerox1212, 2016-08-15:
This a request for someone with a lot of software and UA experience to write a small example to help me out.

As I learn more, I have tried to use UA Modeler to make an object oriented data model for my machine. I now have a decent design, but at this point I have no idea how to connect my python objects to this data model in an efficient way.

Does every matching python object have it's own subscription handler?

Could someone make an example of a small machine which has the python objects matching the UA data model?

If you need some insipration, this is part of my model.
[image]

oroulet, 2016-08-15:
I am not sure I understand your issue. If you have an interval system then
your internal values change, you can call my node.set_value().
If your ua node is the input then you internally subscribe to it and
forward changes to your internal model.

Some SDK provide an api that automatically mirror internal values to ua
nodes. But in the background they do exactly what I described here.

zerox1212, 2016-08-15:
I will make an example, but basically I'm asking this:
Does every Object in the application need a custom SubHandler which maps data to the python object?

zerox1212, 2016-08-15:
You gave me an example a while ago about how to do it. I have tried it and it seems to work, but it basically requires me to make a SubHandler in every python object as a nested class. Is there an easier way? When I did some searching online many stack overflow posts say not to use nested classes in python. I will try to make a quick example tomorrow. Thanks for taking some time to help me out.

oroulet, 2016-08-16:
Yes make an example, You should not need to make nested Class

zerox1212, 2016-08-16:
Here is my example which I tested. It matches part of the model I posted above. (HardwareDevice would be the BeltSpeed custom object type).

```python
class Machine(object):
    """
    test mirroring UA model with python object
    """
    def __init__(self, opcua_server, ua_node):
        self.ParZone0Length = 0
        self.HardwareDevice = None

        class SubHandler(object):
            """
            Subscription Handler. To receive events from server for a subscription
            """

            def __init__(self, machine):
                self.machine = machine

            def datachange_notification(self, node, val, data):
                print("Python: New data change event", node, val, data)
                
                # if one of the subscribed nodes has a data change, find which python attribute to update
                if node == self.machine.test_prop:
                    self.machine.ParZone0Length = data.monitored_item.Value.Value.Value

        # find object properties
        machine_properties = ua_node.get_properties()
        self.test_prop = machine_properties[0]
        self.test_prop.set_value(99.1)
        
        # find children by type
        hw_type = opcua_server.server.nodes.base_object_type.get_child("1:Device")
        machine_hardware = find_your_types(ua_node, hw_type)

        # subscribe to properties
        handler = SubHandler(self)
        sub = opcua_server.server.create_subscription(500, handler)
        handle = sub.subscribe_data_change(machine_properties)

        # create hardware objects for children of type device; each hardware object has its own nested SubHandler class
        self.HardwareDevice = HardwareObject(opcua_server, machine_hardware[0])
        opcua_server.devices.append(self.HardwareDevice)
```

oroulet, 2016-08-16:
In your example there is not reason to have you handler class nested inside the other. move it outside and things will work perfectly

zerox1212, 2016-08-16:
I still need to make a custom SubHandler for every object though, correct? Maybe in the future our modeler can export these things for us, because that will take a lot of time to implement manually. Especially on very complex objects.

oroulet, 2016-08-16:
Can't most of you machines have the same interface? then you can have only one handler class for all.

oroulet, 2016-08-16:
making good generic interfaces to reduce code is important

zerox1212, 2016-08-16:
Well, part of this issue is because I'm more or less using the XML import to configure my application.

Do you know of an easy way to automatically add attributes to my python class based on child nodes? For example if I made a python attribute for every child UA Variable?

I'm not very good with things like building a classes dynamically.

zerox1212, 2016-08-16:
>Can't most of you machines have the same interface? then you can have only one handler class for all.

I can reuse it for the machine object, but other object types need another one. For example my "device" object looks like this:
[image]

So that object needs it's own SubHandler to subscribe to all those properties.

oroulet, 2016-08-16:
That sounds possible with a bit of python magic, use python properties, settattr,getattr, etc

Class X:
      _init__():
            For description in my node.child_descriptions():
                     Settattr(self, description.BrowseName.name, whatever)

zerox1212, 2016-08-16:
I will try that. Maybe then I can have the sub handler loop through all the class attributes instead comparing them explicitly with 20 `elif:` statements.

Thanks.

zerox1212, 2016-08-16:
Thanks to your help I found a decent way to do it.
```python
class SubHandler(object):
    """
    Subscription Handler. To receive events from server for a subscription
    """

    def __init__(self, machine):
        self.machine = machine

    def datachange_notification(self, node, val, data):
        # print("Python: New data change event", node, val, data)

        _node_name = node.get_browse_name()
        setattr(self.machine, _node_name.Name, data.monitored_item.Value.Value.Value)
```
I think I will make a base python class which any object which mirrors a UA object can subclass. The base object with add all children to a dict, then subscribe to all subscribable nodes. Then the above SubHandler will always keep the python object updated and the python object has a dict of it's children in case it needs to write or get more info from UA server.

My class ended up looking like this:
```python
class Machine(object):
    """
    test mirroring UA model with python object
    """
    def __init__(self, opcua_server, ua_node):
        self.nodes = {}
        self.ParZone0Length = 0
        self.HardwareDevice = None

        # keep track of the children of this object (in case python needs to write, or get more info from UA server)
        for _child in ua_node.get_children():
            _child_name = _child.get_browse_name()
            self.nodes[_child_name.Name] = _child

        # find all children which can be subscribed to (python object is kept up to date via subscription)
        sub_children = ua_node.get_properties()
        self.test_prop = sub_children[0]
        self.test_prop.set_value(99.1)
        sub_children.extend(ua_node.get_variables())

        # subscribe to properties/variables
        handler = SubHandler(self)
        sub = opcua_server.server.create_subscription(500, handler)
        handle = sub.subscribe_data_change(sub_children)

        # ADD CUSTOM OBJECT INITIALIZATION BELOW
        # find children by type
        hw_type = opcua_server.server.nodes.base_object_type.get_child("1:Device")
        device_nodes = find_your_types(ua_node, hw_type)

        # create hardware objects for children of type device; each object has its own SubHandler class
        self.BeltSpeed = Device(opcua_server, device_nodes[0])
        opcua_server.devices.append(self.BeltSpeed)
```
Only downside to this system is that every child below the object needs a unique name.

zerox1212, 2016-08-17:
When I'm done with this I will make a new example so new people have an idea of how to have UA objects mirrored by python objects. It might also be useful when we look at auto generating some code from modeler.

closed 2016-08-17 by zerox1212