FreeOpcUa/python-opcua #286: [XML Import] Bug When Importing Variables/Properties

zerox1212, 2016-08-16:
When importing an XML to the server there is a bug that occurs if the XML doesn't specify a "Value" element.

The import parser adds the node with the XML data type, but it doesn't initialize the node with a default value for that data type. As soon as the server starts the node's data type gets set to null. This causes some external clients (such as UA expert) to block writing to the node.

Options to fix:
1. Leave it as is and tell the user that variables/properties with a value of None will become Null data type.
2. If a node is imported with a data type but no value, set the nodes value to an appropriate default to avoid it becoming Null.
3. Always set a value when importing unless XML data type is explicitly set to Null.
4. More of a bug with UA expert, so don't change anything.

Thoughts?

zerox1212, 2016-08-22:
I might go ahead and make a PR for option 2 if no one has an opinion. It's annoying for testing.

oroulet, 2016-08-22:
Does the spec days something about this? Using null seems to make sense....

zerox1212, 2016-08-22:
Honestly I didn't check the spec. Do you know what part to check?

I suppose it does make sense, but it's annoying that you can't write to a node of type null in UA expert.

zerox1212, 2016-08-23:
Thinking about it more I think it's UA modelers fault. If I create a node of type Double, and set the value to 0 and save it. That should mean that I want to set the value to 0 in the XML, not Null. I'll send an email to Unifed Automation and ask them.

zerox1212, 2016-10-18:
I'm still struggling with this and Unified Automation didn't help.
I added a "default values" option to importer. Can we add it to python-opcua?

```python
class XMLParser(object):

    def __init__(self, xmlpath, server, enable_default_values=False):
```

Then this in parser:
```python
def _parse_node(self, name, child):
        """
        Parse a XML node and create a NodeData object.
        """
        ..........

        # set default Value based on type; avoids nodes with Value=None being converted to type Null by server
        if self.enable_default_values:
            if obj.datatype is not None:
                if obj.datatype in ("Int8", "UInt8", "Int16", "UInt16", "Int32", "UInt32", "Int64", "UInt64"):
                    obj.valuetype = obj.datatype
                    obj.value = 0
                elif obj.datatype in ("Float", "Double"):
                    obj.valuetype = obj.datatype
                    obj.value = 0.0
                elif obj.datatype in ("Boolean"):
                    obj.valuetype = obj.datatype
                    obj.value = False
                elif obj.datatype in ("ByteString", "String"):
                    obj.valuetype = obj.datatype
                    obj.value = ""
```