FreeOpcUa/python-opcua #287: [help] Calling a server method with custom inputArguments

Terrub, 2016-08-17:
I originally asked this helprequest in the mailing-list and oroulet instructed me to create an issue out of this for more exposure.
So, by his request, this is the thread:
https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/freeopcua/zwRRI_sQbw4/LjN-FrBWCgAJ

Any and all help would of course be much appreciated.

Right now I could still use some help with setting up the custom class to pass on as an argument.
And if anybody needs testing or intel on getting this functionality properly intergrated, I'll gladly offer my assistance.

Terrub, 2016-08-19:
Quick update. I got it somewhat working using hardcoded values here n there. But I'm sure I can eventually refactor it into something more variable.

I posted a reply with some example code in the maillist post.

zerox1212, 2016-08-23:
I looked at your code. I don't know much about extension object, but I agree with @oroulet last post about how to find the hierarchy.  From what you said about the server people being surprised about the XML dictionary I'm wondering if that XML in the encoding node description is just by chance. It sounds like some obscure spec that any custom structure should be described in XML. If the creator of the server generated their UA model, they probably didn't even realize it made that XML for them.

Terrub, 2016-08-24:
It does indeed seem to be standard OPC UA behaviour. As far as I understand they created these custom structures using certified UA tools ([UAModeller](https://www.unified-automation.com/products/development-tools/uamodeler.html) I believe)
They are using a different stack though, in case that makes a difference, although I hope not.

As said, I can find an example of the xml structure in the book I mentioned. (Page 67 fig. 2.34)

I haven't had a chance to test the method @oroulet mentioned yet but I am going to later this week. Sadly I'm multi-tasking different things atm.
I'll let you know what I find but I'm not entirely sure this will work as I'm looking at a structure here, not a set of nodes, so ordering by NodeId is going to be interesting.

oroulet, 2016-08-24:
I may be wrong here since we have not implemented struts yet. Maybe the spec requires to define structure order in xml with a hasEncoding reference...... Let me know

Terrub, 2016-08-24:
Alright I browsed through the specs a little and I found two rather interesting things.
Mind you I'm referring to the specifications found over at https://opcfoundation.org/developer-tools/specifications-unified-architecture and you need to have an account to view them. For now I'm going to assume that you have used these specs to build what you have already.

So the two parts I've found that are interesting are:
- Part 3: Address Space Model
-- Specifically the section on **OPC Binary Type Description System**
- Part 6: Mappings
-- Specifically the sections on **ExtensionObject**

The XML that I get from the server is this **OPC Binary Type Description** and apparently is what is used to decypher how to decode binary encoded objects like the custom input arguments and event data that I had trouble with.
I'm guessing that you'd need to load up the XML and use that as a dictionary to interpret the data described in the structures.

Hope this helps.

zerox1212, 2016-08-24:
Thanks for the investigation. Maybe sometime in the future we can add real support for extension objects.

SeetaiahB, 2017-07-12:
Question on ExtensionObject:

I need to access/use GetData Method and SetData Method. Just wonder how to phrase input and out arguments for particular method call. Appreciated if you could send some example code with these methods.

Another question is on Extension object.
When I executed below command print(client.get_node(ua.NodeId(5069,2)).call_method(ua.NodeId(7073,2)))

It prints a Extension object
ExtensionObject(TypeId:FourByteNodeId(ns=2;i=5079), Encoding:1, 160 bytes)

When I use get_value() method on above object it thrown an error. Just wonder how to access the data of 160Bytes information from above extension object.