FreeOpcUa/python-opcua #288: [help] Block Subscription Callbacks

zerox1212, 2016-08-19:
My application needs to work on some data held in multiple objects. These objects mirror UA objects and are kept in sync via subscription.

What is the best way to block/ignore the subscription callback? I need to work on the data in these objects without the subscription thread changing the data during doing some heavy calculation loops.

oroulet, 2016-08-19:
In theory it is possile to pause subscriptions but this is not implemented.
I think a correct solution is to take a snapshot of your values and use that for computation.

oroulet, 2016-08-19:
btw if you change things in exporter or modeller, make a PR asap so we do not make double work or get issues merging

zerox1212, 2016-08-19:
If I start anything I will make a PR.

zerox1212, 2016-08-22:
I will manually copy the values i'm working on to avoid this issue. In the future it would probably good to implement this. Especially if we want to generate python objects from UA objects and keep the synchronized.

zerox1212, 2016-08-23:
I have a new related question. When using the server do internal subscriptions obey the publishing interval?

closed 2018-01-25 by zerox1212