FreeOpcUa/python-opcua #290: get_data_value() - question

jamalpatel, 2016-08-22:
I'm very new to OPC and your project. I used get_data_value() to get value from the node, which works great, but having problem parsing the return. 
Is there any way to parse the return to Dictionary data type? Is there a helper function that could parse the return?

zerox1212, 2016-08-22:
What do you want a dictionary of? The DataValue object? Why do you need a dict? you can just work with the DataValue object directly.

jamalpatel, 2016-08-22:
How can I work with DataValue object to directly? My return looks like this, and I need to get to the values of specific array's element:

DataValue(Value:Variant(val:[DataValue(Value:Variant(val:[Variant(val:1,type:VariantType.UInt16), Variant(val:17.08,type:VariantType.String), Variant(val:6.00999975204,type:VariantType.Float), Variant(val:2.16600012779,type:VariantType.Float), Variant(val:0.0883099287748,type:VariantType.Float), Variant(val:429.799987793,type:VariantType.Float)],type:VariantType.Variant), StatusCode:StatusCode(Good), SourceTimestamp:2016-08-17 11:01:28, ServerTimestamp:2016-08-22 14:01:40)],type:VariantType.DataValue), StatusCode:StatusCode(Good), SourceTimestamp:2016-08-22 14:01:40, ServerTimestamp:2016-08-22 14:01:40)

zerox1212, 2016-08-22:
The values are inside your object. To be specific, they are returned as a variant. This is another object specified by OPC UA to describe your value in detail (such as what type it is).

I don't work with arrays, but normally to get to just a value you can just do something like this.
```my_dv = node.get_data_value().Value.Value```

Since you have an array, your DataValue.Value is probably a list. so you will need to specify which array member you want.
```my_dv = node.get_data_value().Value[0].Value```

Or if you work with the entire data value object its like this.
```my_dv = node.get_data_value()
some_value = my_dv.Value[0].Value```

jamalpatel, 2016-08-22:
Perfect, I will give it a try. Thank you

zerox1212, 2016-08-22:
I had the some problem when I started. I even have a screen shot of when I was learning lol.
[image]

jamalpatel, 2016-08-22:
It works like a charm. Thanks a lot for your help.

closed 2016-08-22 by jamalpatel