FreeOpcUa/python-opcua #292: improve export of values, support list and extension objects

oroulet, 2016-08-22:
I am looking at that

zerox1212, 2016-08-23:
I don't have time to work on the exporter for a while. Too much work for my employer at the moment.

SeetaiahB, 2017-07-12:
Question on ExtensionObject:

I need to access/use GetData Method and SetData Method. Just wonder how to phrase input and out arguments for particular method call. Appreciated if you could send some example code with these methods.

Another question is on Extension object.
When I executed below command print(client.get_node(ua.NodeId(5069,2)).call_method(ua.NodeId(7073,2)))

It prints a Extension object
ExtensionObject(TypeId:FourByteNodeId(ns=2;i=5079), Encoding:1, 160 bytes)

When I use get_value() method on above object it thrown an error. Just wonder how to access the data of 160Bytes information from above extension object.

oroulet, 2017-07-12:
https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client_read-custom_structures.py

oroulet, 2017-07-12:
and please do not post the same question everywhere

SeetaiahB, 2017-07-12:
I am sorry I was accidently done …thanks

SeetaiahB, 2017-07-13:
hi orulet,

when i used they presented way 
https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client_read-custom_structures.py

the following error message displayed.
registring new extension object: Amps FourByteNodeId(ns=2;i=5029) <class 'structures_Elvs.Amps'>
registring new extension object: BasicMetering FourByteNodeId(ns=2;i=5086) <class 'structures_Elvs.BasicMetering'>
registring new extension object: BreakerSettings FourByteNodeId(ns=2;i=5079) <class 'structures_Elvs.BreakerSettings'>
registring new extension object: OCSettings FourByteNodeId(ns=2;i=5058) <class 'structures_Elvs.OCSettings'>
registring new extension object: PkUp_Delay FourByteNodeId(ns=2;i=5050) <class 'structures_Elvs.PkUp_Delay'>
registring new extension object: Rating FourByteNodeId(ns=2;i=5041) <class 'structures_Elvs.Rating'>
registring new extension object: RelaySettings FourByteNodeId(ns=2;i=5025) <class 'structures_Elvs.RelaySettings'>
registring new extension object: Volts FourByteNodeId(ns=2;i=5031) <class 'structures_Elvs.Volts'>
Traceback (most recent call last):
  File "C:/Projects/2017/Entellisys6.0_CPU_Testing/Functions_Library/CPU_Functions/OPC_Functions.py", line 47, in <module>
    after = struct.get_value()
  File "C:\Projects\2017\Entellisys6.0_CPU_Testing\CPUTesting_3.5Virtual_Envi\lib\site-packages\opcua\common\node.py", line 129, in get_value
    result = self.get_data_value()
  File "C:\Projects\2017\Entellisys6.0_CPU_Testing\CPUTesting_3.5Virtual_Envi\lib\site-packages\opcua\common\node.py", line 138, in get_data_value
    return self.get_attribute(ua.AttributeIds.Value)
  File "C:\Projects\2017\Entellisys6.0_CPU_Testing\CPUTesting_3.5Virtual_Envi\lib\site-packages\opcua\common\node.py", line 245, in get_attribute
    result[0].StatusCode.check()
  File "C:\Projects\2017\Entellisys6.0_CPU_Testing\CPUTesting_3.5Virtual_Envi\lib\site-packages\opcua\ua\uatypes.py", line 207, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadAttributeIdInvalid: The attribute is not supported for the specified Node.(BadAttributeIdInvalid)

Appreciated if you could share some examples to call extension object retrieve information like body etc..from it.

zerox1212, 2017-07-13:
Post your code and example of extensionobject (and encoding).

SeetaiahB, 2017-07-13:
Here is the my code:

[cid:image001.png@01D2FBFD.D01AF150]            [cid:image002.jpg@01D2FBFD.D01AF150]  [cid:image003.png@01D2FBFD.D01AF150]
Node structure                                                                 Output Argument Attributes for get Data                               SetData Input Argument

Above is the my server node structure pulled from UaExpert.

I need to access/use GetData Method and SetData Method. Just wonder how to phrase input and out arguments for particular method call. Appreciated if you could send some example code with these methods.

Another question is on Extension object.

When I executed below command print(client.get_node(ua.NodeId(5069,2)).call_method(ua.NodeId(7073,2)))

It prints a Extension object
ExtensionObject(TypeId:FourByteNodeId(ns=2;i=5079), Encoding:1, 160 bytes)

When I use get_value() method on above object it thrown an error. Just wonder how to access the data of 160Bytes information from above extension object.

Appreciated your help

mkdthanga, 2022-09-08:
I get this following error: 
`opcua.ua.uaerrors._auto.BadAttributeIdInvalid: "The attribute is not supported for the specified Node."(BadAttributeIdInvalid)`
Any one knows what it says and how to fix it? 

I want to read the complete structure of a node and I want to export all its node IDs and store it into a CSV file, so that it would be easy to import it in Nodered!

This is the code I used:
```
`import sys
sys.path.insert(0, "..")
import time
import logging
from IPython import embed

from opcua import Client
from opcua import ua

if __name__ == "__main__":
    logging.basicConfig(level=logging.WARN)
    client = Client("opc.tcp://192.168.64.194:4840")
    try:
        client.connect()
        root = client.get_root_node()
        objects = client.get_objects_node()
        struct = client.get_node("ns=2;i=5001")
        struct_array = client.get_node('"ns=3;s="Control"')
        before = struct.get_value()
        before_array = struct_array.get_value()
        client.load_type_definitions()  # scan server for custom structures and import them
        after = struct.get_value()
        after_array = struct_array.get_value()
        

        embed()
    finally:
        client.disconnect()`
```