FreeOpcUa/python-opcua #293: [help] Retrieving event data with custom objects

Terrub, 2016-08-23:
This is related to my earlier question that now works using hardcoded values, but is not at all dynamic just yet.

I've got the inverse problem now that the call I make returns amongst other things, a custom object. This object is described in the same XML-dictionary document that describes the method. Though I should be able to retrieve the description dictionary if it wasn't, I haven't looked into it.

The code below is what I do now, but it feels like this is not at all the right way to do this.
So obviously the question is: What is the right way to do this?

```
def unpack_custom_data(p_custom_data):
    data = utils.Buffer(p_custom_data)
    measurements = []

    name = ua.unpack_uatype("CharArray", data)
    type = ua.unpack_uatype("Int32", data)
    num_measurements = ua.unpack_uatype("Int32", data)
    for idx in range(0, num_measurements):
        measurement = ua.unpack_uatype("Float", data)
        measurements.append(measurement)
    percentage = ua.unpack_uatype("Float", data)
```

zerox1212, 2016-08-23:
I don't really understand. Now that you can call the server method with custom objects you need to unpack what it returns?

Terrub, 2016-08-24:
I believe so. One of the returned values is another custom structure that needs to be decoded from a bytestring to something digestible. It is that value (or the Body actually) of that ExtensionObject that I decode. I should probably create a custom class and put this logic in the `from_binary` method but I am not sure I'm doing this the right way to start with.

oroulet, 2016-08-24:
Start by writing a simple function, then we look at where to place it

oroulet, 2016-08-24:
event start with a simple function with anode as arg and return the xml encoding. So we can test this different places. From you email the xml format representing the encoding sems the same one as the one used to generate the address space, so we can copy code from it later

oroulet, 2016-08-24:
The following code describes roughly what I do:

First I get the list of arguments from the method using the inputarguments node:

```
input_argument_node = method_node.get_child(["0:InputArguments"])
arguments = input_argument_node.get_value()
```

Then I check if the argument has an encoding associated.

for argument in arguments:
…

```
    encodings = opc_ua_client.get_node(argument.DataType).get_references(
        refs=ua.ObjectIds.HasEncoding,
        direction=ua.BrowseDirection.Forward
    )
```

If it has an encoding associated, I grab one of the options and get the node ID from it:

```
chosen_encoding = encodings[0]
encoding_node = opc_ua_client.get_node(chosen_encoding.NodeId)

```
This was the first point where I thought I had done goofed, but this list of encodings is exactly 1, There is only 1 encoding as far as I can tell.

Moving on, I get the description of the encoding:
   
```
data_type_description = encoding_node.get_references(
    refs=ua.ObjectIds.HasDescription,
    direction=ua.BrowseDirection.Forward
)
data_type_description_node = opc_ua_client.get_node(data_type_description[0].NodeId)

```
This was the second point where I thought I done goofed, but once again, list length is 1. There is only 1 description of this encoding (which would make sense)

From here I can do an inverse search to get the dictionary where the description is located in.

```
data_type_description_dictionaries = data_type_description_node.get_references(
    refs=ua.ObjectIds.HasComponent,
    direction=ua.BrowseDirection.Inverse
)
```

Then I can select one of the references (Of which there was, again, only 1) get the node and read the contents:

```
dictionary_reference = data_type_description_dictionaries[0]
xml_dictionary = opc_ua_client.get_node(dictionary_reference.NodeId).get_value()

```
This xml_dictionary will contain the XML I was referring to.

oroulet, 2016-08-24:
```
<opc:StructuredType BaseType="ua:ExtensionObject" Name="DeviceConfigurationType">
  <opc:Field TypeName="opc:CharArray" Name="label"/>
  <opc:Field TypeName="ns1:SettingTypeDefinition" Name="SettingType"/>
  <opc:Field TypeName="opc:Float" Name="value1"/>
  <opc:Field TypeName="opc:Double" Name="value2"/>
</opc:StructuredType>
```

oroulet, 2016-08-24:
a shorter version of the code above:

`
encs = data_type.get_children(refs=ua.ObjectIds.HasEncoding)
desc = encs[1].get_children(refs=ua.ObjectIds.HasDescription)
dic = desc[0].get_references(
    refs=ua.ObjectIds.HasComponent,
    direction=ua.BrowseDirection.Inverse
)
xml_dictionary = client.get_node(dic.NodeId).get_value()
`
The issue here is that the node "dic" is just 0:Root\0:Types\0:DataTypes\0:Opc.Ua" and
in my caseit contains just the address of the xml description spec:
b'<opc:TypeDictionary\r\n  xmlns:opc="http://opcfoundation.org/BinarySchema/"\r\n  xmlns:xsi="http://www.w3.org/2001/XML'

Did they put the entire spec in the node value on your server? also containing your custom structure? I do not think this is part of spec.....

oroulet, 2016-08-24:
Or did they only add the definiton of their custom structures?

Terrub, 2016-08-24:
How did you find out the path from root for the dictionary node? Is that knowledge of the system or can you request that from the server in some manner? Because I'd like to know if it is the same for my situation.

I do believe they added the definition of the custom structures, the question is where is it located.
It would seem rather silly if the book explained such an elaborate method of getting to the dictionary containing the descriptions of the custom structures if there is only 1 node that contains all the descriptions in the first place. Also I believe the book refers to it as one amongst many so I'm guessing that each data structure has it's own list of things.

I received a zip with the modelling data they used, I'll have a look if there are any xml files in that zip and if they match the xml that I receive from the server.

oroulet, 2016-08-24:
> How did you find out the path from root for the dictionary node?

Good point! I added a method get_current_path() to master just now

> I do believe they added the definition of the custom structures, the question is where is it located.

windering the same thing. It looks like the node structure does not exist for data types under BaseDataTypes/Structure

Terrub, 2016-08-24:
I've had a look at the actual address space and the XML that I get seems to be a sibling of the OPC.Ua node that you found @ 0:Root\0:Types\0:DataTypes\0:Opc.Ua
I'd have to update my stack and check the path with your `get_current_path()` method to see if it is as I expect (i.e. something similar to "0:Root\0:Types\0:DataTypes\5:Custom_Stuff")

Does that help?