FreeOpcUa/python-opcua #297: [Server/Noob Question] How do I get methods to work in my custom types?

stillero, 2016-08-30:
Hello!

I recently started working on OPC UA and Python, so I'm still trying to wrap my head around most things, so please, bear with me. This might be a pretty stupid question, but I just can't get it right, so I need to ask anyway - how do I get methods to work in my custom types?

Here's a short the example, based on the "server-minimal" example:

[image]

Whenever I call the method via Test Client, it always returns "BadNothingToDo".
But I know this method works just fine! If I add it to an object, like the SmoothSample above, the method runs successfully when called by the client. But that can't be the only solution, right?
I know I'm doing something wrong...

Please help me out!

Daniel

zerox1212, 2016-08-30:
You are creating a method node that doesn't have a UaMethod. You need to use a decorator when defining ```SmoothMethod1```

You can see this in server-method example. https://github.com/FreeOpcUa/python-opcua/blob/master/examples/server-methods.py
```python
# method to be exposed through server
def func(parent, variant):
    ret = False
    if variant.Value % 2 == 0:
        ret = True
    return [ua.Variant(ret, ua.VariantType.Boolean)]

# method to be exposed through server
# uses a decorator to automatically convert to and from variants

@uamethod
def multiply(parent, x, y):
    print("multiply method call with parameters: ", x, y)
    return x * y
```

stillero, 2016-08-30:
Thanks for the Input, I've tried your solution, but still no change in the outcome. Did I implement it wrong?

[image]

zerox1212, 2016-08-30:
Your second attempt is wrong, because you have to declare method stuff manually.

```python
inargx = ua.Argument()
    inargx.Name = "x"
    inargx.DataType = ua.NodeId(ua.ObjectIds.Int64)
    inargx.ValueRank = -1
    inargx.ArrayDimensions = []
    inargx.Description = ua.LocalizedText("First number x")
    inargy = ua.Argument()
    inargy.Name = "y"
    inargy.DataType = ua.NodeId(ua.ObjectIds.Int64)
    inargy.ValueRank = -1
    inargy.ArrayDimensions = []
    inargy.Description = ua.LocalizedText("Second number y")
    outarg = ua.Argument()
    outarg.Name = "Result"
    outarg.DataType = ua.NodeId(ua.ObjectIds.Int64)
    outarg.ValueRank = -1
    outarg.ArrayDimensions = []
    outarg.Description = ua.LocalizedText("Multiplication result")
```
However I just tested it and it doesn't work. Need to wait for @oroulet or @destogl, they know a lot more about how methods work.

zerox1212, 2016-08-30:
After comparing the working methods in the example with the one added to the object type something isn't right.

I don't have any more time to help you at the moment, but you could look at https://github.com/FreeOpcUa/python-opcua/blob/master/examples/server-instantiate-object.py and try to call ```instantiate()``` to create an instance of your ```SmoothType```. Perhaps this will create the method correctly.

stillero, 2016-08-30:
Thanks for responding so quickly! Unfortunately, still the same outcome - SmoothMethod1 returns BadNothingToDo and SmoothMethod2 works. I'm sorry if I failed again at implementing it correctly, I really appreciate your help @zerox1212!
[image]

oroulet, 2016-08-31:
at first look you code is OK. but the examples and tests here work.
Can you paste as code and not as image and I may try it
What client do you use? what is client code?

stillero, 2016-08-31:
Sure thing, thanks for taking a look at it!

I'm using the UA Expert Client from Unified Automation just to test connectivity and to try and call my methods.

[smooth.zip](https://github.com/FreeOpcUa/python-opcua/files/446537/smooth.zip)

oroulet, 2016-08-31:
Does the example from examples first work on your pc?I just tested it with
uaexpert and it works. What do you change in yours?

stillero, 2016-08-31:
Yeah, the examples all work just fine, but when I assign the same method to a custom object-type, then create an object-node with my custom object-type, and use uaexpert to call the embedded method, it returns "BadNothingToDo"

[image]

And if I assign the same method later on to the Object-node again, it works. I just dont get it.

[image]

oroulet, 2016-08-31:
I just tested a slightly modified version and it works:

@uamethod
def smooth(parent, var):
    ret = False
    if var % 2 == 0:
        ret = True
    return ret

if __name__ == "__main__":

    # setup our server
    server = Server()
    server.set_endpoint("opc.tcp://0.0.0.0:4840/freeopcua/server/")

    # setup our own namespace, not really necessary but should as spec
    uri = "http://examples.freeopcua.github.io"
    idx = server.register_namespace(uri)

    # get Objects node, this is where we should put our nodes
    objects = server.get_objects_node()

    types = server.get_node(ua.ObjectIds.BaseObjectType)
    SmoothType = types.add_object_type(idx, "SmoothType")

    SmoothType.add_method(idx, "SmoothMethod1", smooth, [ua.VariantType.Int64], [ua.VariantType.Boolean])

    # starting!
    server.start()`
`

Does that one works?

oroulet, 2016-08-31:
instanciation og methods might not work, I have not tested it,

stillero, 2016-08-31:
That one doesn't work for me either. Could it be, that I implemented the sdk wrong?

oroulet, 2016-08-31:
what version of opcua lib? But this has worked for a long time ....

stillero, 2016-08-31:
current version is 0.10.14

zerox1212, 2016-08-31:
I had the same problem as him. Could it be a Python 3.5 problem? What version of python are you using @oroulet .

oroulet, 2016-08-31:
works with both python 2.7 and 3.5- what version of uaexpert? I have 1.4.2

But do you say that the example works but not the code I posted in the bug report? Then you should be able to spot the difference and find out what breaks

zerox1212, 2016-08-31:
The code you posted only creates the type. You are not making a new object which uses your type.

oroulet, 2016-08-31:
isnt't it the goal? As I said instanciation of a method does not work. code for it is missing.
I am not sure how this should be implemented. Should the  instantiated method call the original method function? or should we offer some kind of API to easily link the new method to a python function.... not sure...

zerox1212, 2016-08-31:
Here is my full test code which reproduces what he sees. Try to call the method in my_obj.

```python
import sys
sys.path.insert(0, "..")
import logging

try:
    from IPython import embed
except ImportError:
    import code

    def embed():
        vars = globals()
        vars.update(locals())
        shell = code.InteractiveConsole(vars)
        shell.interact()

from opcua import ua, uamethod, Server, instantiate

# method to be exposed through server
def func(parent, variant):
    ret = False
    if variant.Value % 2 == 0:
        ret = True
    return [ua.Variant(ret, ua.VariantType.Boolean)]

# method to be exposed through server
# uses a decorator to automatically convert to and from variants

@uamethod
def multiply(parent, x, y):
    print("multiply method call with parameters: ", x, y)
    return x * y

if __name__ == "__main__":
    # optional: setup logging
    logging.basicConfig(level=logging.WARN)
    #logger = logging.getLogger("opcua.address_space")
    # logger.setLevel(logging.DEBUG)
    #logger = logging.getLogger("opcua.internal_server")
    # logger.setLevel(logging.DEBUG)
    #logger = logging.getLogger("opcua.binary_server_asyncio")
    # logger.setLevel(logging.DEBUG)
    #logger = logging.getLogger("opcua.uaprocessor")
    # logger.setLevel(logging.DEBUG)
    #logger = logging.getLogger("opcua.subscription_service")
    # logger.setLevel(logging.DEBUG)

    # now setup our server
    server = Server()
    #server.set_endpoint("opc.tcp://localhost:4840/freeopcua/server/")
    server.set_endpoint("opc.tcp://0.0.0.0:4840/freeopcua/server/")
    server.set_server_name("FreeOpcUa Example Server")

    # setup our own namespace
    uri = "http://examples.freeopcua.github.io"
    idx = server.register_namespace(uri)

    # get Objects node, this is where we should put our custom stuff
    objects = server.get_objects_node()

    # populating our address space
    myfolder = objects.add_folder(idx, "myEmptyFolder")
    myobj = objects.add_object(idx, "MyObject")
    myvar = myobj.add_variable(idx, "MyVariable", 6.7)
    myvar.set_writable()    # Set MyVariable to be writable by clients
    myarrayvar = myobj.add_variable(idx, "myarrayvar", [6.7, 7.9])
    myarrayvar = myobj.add_variable(idx, "myStronglytTypedVariable", ua.Variant([], ua.VariantType.UInt32))
    myprop = myobj.add_property(idx, "myproperty", "I am a property")
    mymethod = myobj.add_method(idx, "mymethod", func, [ua.VariantType.Int64], [ua.VariantType.Boolean])

    inargx = ua.Argument()
    inargx.Name = "x"
    inargx.DataType = ua.NodeId(ua.ObjectIds.Int64)
    inargx.ValueRank = -1
    inargx.ArrayDimensions = []
    inargx.Description = ua.LocalizedText("First number x")
    inargy = ua.Argument()
    inargy.Name = "y"
    inargy.DataType = ua.NodeId(ua.ObjectIds.Int64)
    inargy.ValueRank = -1
    inargy.ArrayDimensions = []
    inargy.Description = ua.LocalizedText("Second number y")
    outarg = ua.Argument()
    outarg.Name = "Result"
    outarg.DataType = ua.NodeId(ua.ObjectIds.Int64)
    outarg.ValueRank = -1
    outarg.ArrayDimensions = []
    outarg.Description = ua.LocalizedText("Multiplication result")

    #multiply_node = myobj.add_method(idx, "multiply", multiply, [inargx, inargy], [outarg])

    my_type = server.nodes.base_object_type.add_object_type(idx, "my_type")
    my_type.add_method(idx, "multiply", multiply, [inargx, inargy], [outarg])
    myobj1 = instantiate(server.nodes.objects, my_type, bname="2:myobj1")

    # starting!
    server.start()
    print("Available loggers are: ", logging.Logger.manager.loggerDict.keys())
    try:

        embed()
    finally:
        server.stop()

```

oroulet, 2016-08-31:
yes you instantiate and instatiation of a method results in a dead method returning BadNothingToDo. A new function must be linked to the node. We can easiy add such a method. maybe at the node level Node.link_function(my_function)

stillero, 2016-08-31:
I think I failed to communicate to you what my intend was, but you found out yourselves. What I wanted to do was creating a custom object type with a method so that when I create an object-node of said type, the object-node already has the method embedded. 

Here's an example how I had it in mind:

Say we have a pressure switch. I would like to model a object-type PressureSwitchType which already contains a variable-node "status" that holds a boolean True/False, depending on if the switch is pressed at the moment.

Now let's say I have a shelf that holds boxes like a magazine. You take a box out, the rest of the boxes slide down/move up. A shelf is equipped with 2 pressure switches. One at each end of the shelf. If both are pressed at the same time, one can say, the shelf is full and can't store another box.

So, I would like to model a obcet-type ShelfType which has two object-nodes of the type PressureSwitchType, a variable-node "capacity" and a method-node that checks for said capacity level.

Now imagine I don't have just one shelf instance, but 500. It would be really inconvenient if I were to add a method 500 times, right? 

And if it was more complex then this example, it would be even more inconvenient, wouldn't it?

Or am I wrong? Am I trying to implement it is not supposed to in OPC UA? Is there a smarter way to model such a use case? That is, regardless of actual use, this was just an example.

oroulet, 2016-08-31:
as I said I did not know how to implement this. but looks like you have a good point for allowing a method to follow instantiation. Thinking about it I am not sure it is possible since instantiate uses only standard ua methods and this has not reference to the internal server function.

But it still possible to automate. this. just get method node after instantiation and map a method to it (might be the same for all methods, and use the parent to change target device)

inode = instantiate(xxxxx)
instantiated_method = inode.get_child("1:mymethod")
instantiate_method.link_function(my_function)  # this one is currently missing, must be added, feel free to have a look. Look at implementation of add_method

oroulet, 2016-08-31:
In fact, do not put me method in Node class. This is server specific, put it on server object

zerox1212, 2016-09-29:
I started looking at how to add this feature. It doesn't seem too difficult, however I'm wondering if it's worth adding a module which holds all user methods. Then on instantiation we can search user_methods.py for the function instead of explicitly giving the instantiated method a python function.

Thoughts?

zerox1212, 2016-09-29:
So it turns out solving this issue with explicit assignment is easy. Just add this to server.py and call it after instantiation as @oroulet said.
```python
def link_method(self, node, callback):
        node.server.add_method_callback(node.nodeid, callback)
```

Solving this automatically is a bit harder because ```instantiate()``` doesn't have access to the server at the moment.

Is it worth to make a PR to do something like this?
instantiate.py
```python
def instantiate(server, parent, node_type, nodeid=None, bname=None, idx=0):
    """
    instantiate a node type under a parent node.
    nodeid and browse name of new node can be specified, or just namespace index
    If they exists children of the node type, such as components, variables and 
    properties are also instantiated
    """
    rdesc = _rdesc_from_node(parent, node_type)

    if nodeid is None:
        nodeid = ua.NodeId(namespaceidx=idx)  # will trigger automatic node generation in namespace idx
    if bname is None:
        bname = rdesc.BrowseName
    elif isinstance(bname, str):
        bname = ua.QualifiedName.from_string(bname)    

    nodeids = _instantiate_node(parent.server, parent.nodeid, rdesc, nodeid, bname)

    instantiated_nodes = []
    for nid in nodeids:
        inst_node = Node(parent.server, nid)
        if inst_node.get_node_class() == ua.NodeClass.Method:
            server.link_method(inst_node)

        instantiated_nodes.append(inst_node)
    return instantiated_nodes
```
server.py
```python
    def link_method(self, node, callback=None):
        if callback:
            node.server.add_method_callback(node.nodeid, callback)
        else:
            func_name = node.get_browse_name().Name
            node.server.add_method_callback(node.nodeid, getattr(user_methods), func_name)
```

Then user can put all their custom methods in user_methods.py and instantiate will connect them by name.

zerox1212, 2016-10-10:
Fixed with https://github.com/FreeOpcUa/python-opcua/issues/313 for now. Should be discussed in the future. OPC UA spec does have a few notes about methods, mainly that methods which are part of an object type should act like a static method.

This could be automated as described above, but for now is up to the user. I think this can be closed...

destogl, 2019-07-31:
Is this solved? It seems so...

zerox1212, 2019-07-31:
It can be closed, but know that any time an user calls instantiate, they must manually link methods of what they are instantiating. There is nothing automatic.

closed 2019-08-02 by destogl