FreeOpcUa/python-opcua #298: ServerTimestamp returns as NoneType

LBeaufore, 2016-09-02:
When I attempt to access the ServerTimestamp of a node by using
`node.get_data_value().ServerTimestamp`
A "NoneType" value is returned. However, SourceTimestamp, val, and StatusCode seem to work in the place of ServerTimestamp. Is ServerTimestamp still supported in this way?

oroulet, 2016-09-03:
Some servers do no set ServerTimestamp attribute

LBeaufore, 2016-09-06:
I am almost certain this one does, it returned a server timestamp with Unified Automation's UaExpert GUI Client

oroulet, 2016-09-06:
There's been an issue on this a few month ago, maybe try to update.
Otherwise you need to start wireshark and compare a session against python
and one against uaexpert and try to spot difference, not very difficult
since wireshark had built in support for opcua

LBeaufore, 2016-09-06:
That makes sense, I'll try it out. Thank you!

sanlop73, 2017-11-17:
Good afternoon:
Could you please give me the solution?
Getting the same issue here, Attached picture of UAexpert compare with
            print("SourceTimeStamp", val1.get_data_value().SourceTimestamp)
            print("ServerTimeStamp", val1.get_data_value().ServerTimestamp)

Thanks in advance

[image]

oroulet, 2017-11-17:
@sanlop73 record a wireshark session. you can also make use that we ask for both timestamps in query. I kind of remember this is part of read query

sanlop73, 2017-11-20:
Good morning:
I have made your suggested test with Wireshark.
By default the library is asking for SourceTimestamp only, for read.
So i have modified the file Node.py
In function get_attribute, i have add 
**params.TimestampsToReturn = ua.TimestampsToReturn.Both**

Is this the right function to place the instruction?

Thanks in advance for your comments.

[image]
[image]

sanlop73, 2017-11-20:
Sorry , forgot to upload the image

[image]

oroulet, 2017-11-20:
correct, It might be that we should have "both" as default

sanlop73, 2017-11-20:
Yes, please, is better, you always have the chance to call or not to the variable.
Thanks