FreeOpcUa/python-opcua #304: migrate browsename namespace when importing to correct index

oroulet, 2016-09-22:


oroulet, 2016-09-22:
@maljac do you remember why the nodeid index migration is in xmlparser.py and not xmlimporter.py? looking at code I think it should have been written in xmlimporter.py but maybe you had a reason...
I need to implement migration of browsename index for opcua.modeller....

maljac, 2016-09-22:
If I remember right, it was mainly because we already iterate over all elements of the model here, and because we have the original namespace id (the namespace id defined in the xml).
Looking at the code, the namespace's get created in the xmlparser too (https://github.com/FreeOpcUa/python-opcua/blame/master/opcua/common/xmlparser.py#L94) which seems like a inappropriate place.

I think I did it because you need the namespace id defined in the xml. If you do not do it here you might have to introduce another object for the NamespaceURIs which have to keep theire order as defined in the xml or get a ID (since the order of the NamespaceURIs define theire relation to the namespace IDs of the nodes defined in XML model). Then in the XMLImporter you would actually create the namespaces and replace the original namespace id for all node / parent node referrences:

bitkeeper, 2016-10-10:
I hit the same problem when importing multiple namespaces.
Fixed proposed in PR #322 all due it is fixed in xmlparser.py and not xmlimporter.py.

bitkeeper, 2016-10-25:
@oroulet I think this issue can be closed, due it is fixed ?

closed 2016-10-25 by oroulet