FreeOpcUa/python-opcua #306: Stopping/Interrupting with Ctrl+C

OptrixAU, 2016-09-23:
After you've called 'start' on your OPC server, my program doesn't appear to respond to KeyboardExceptions anymore (ie. pressing Ctrl+C to close the application).

I've gotten around this by adding signal handlers that allow me to stop the server as needed.

Should the server respond to a Keyboard or SysExit exception by closing its thread? Or is the existing behaviour expected?

Otherwise, I'm extremely impressed by how easy this is - amazing work!

oroulet, 2016-09-23:
I use ctrl-c all the time, so it definitely works out of the box. But we
use internally asyncio, and I have heard that this library sometimes break
things for users having special signal handlers in their setup

OptrixAU, 2016-09-23:
My main thread is I/O blocked - perhaps the combination of the two causes some issues.

oroulet, 2016-09-23:
That the main thread is blocked should not be an issue. Everything runs in
background threads

zerox1212, 2016-09-23:
I had similar problems. https://github.com/FreeOpcUa/python-opcua/issues/69

My research concluded that asyncio breaks exiting functions. My advice is to find a way to call server stop explicitly.

oroulet, 2016-09-23:
Server. Stop() MUST be called. Of course if you do not stop it the
application does not exist

OptrixAU, 2016-09-24:
Hmmm...it looks like the CTRL+C exception is only thrown on the main thread,  and requests/curl can eat the exception.

Adding a signal handler allows me to stop OPC UA and also stop the Request loop, although because the Request loop is on the main thread,  I have to wait for the request to complete/fail/timeout first.

oroulet, 2016-09-24:
I think there is a known issue with asyncio. So Google for it for the
details

oroulet, 2016-09-24:
But call stop!

OptrixAU, 2016-09-27:
OK - the primary cause of the issue was NOT FreeOPCUA, but it was the fact that my main thread was blocked by other I/O tasks too, which meant that I wasn't getting the KeyboardException in a timely manner.

I solved the problem simply by pushing that additional I/O off to an extra thread, leaving the main thread in a sleep-loop.

Since it's the main thread that receives the exception, adding a signal handler on the main thread allowed me to catch the CTRL+C very quickly (you often get an I/O exception instead of a KeyboardException, as the call to 'sleep' gets interrupted), and gracefully shut down both my own I/O thread and the FreeOPCUA server.

Any admins should mark this as 'Not A Bug' and close it - although I wouldn't mind seeing something like this shown in the basic examples, since I'd imagine that I'm not alone in needing to do something like this.

destogl, 2016-09-27:
Can you add this into example? You add it as comment with description.

hunterjackson, 2016-12-19:
Ctrl + \ works when Ctrl + c doesn't work for me