FreeOpcUa/python-opcua #307: AttributeError: module 'opcua.ua' has no attribute 'pack_bytes'

AlexejStukov, 2016-09-23:
`/ua/uatypes.py` was reworked heavily in this [commit](https://github.com/FreeOpcUa/python-opcua/commit/3976fb13a7b5fc2e881f6a3fea2e8a60a9754eb3) and that broke the client, because `pack_bytes` was removed.

oroulet, 2016-09-23:
All tests pass, do this is a special case. Can you give more information?

AlexejStukov, 2016-09-23:
It's not really a special case, just using a password. `Client._encrypt_password` is still using the removed `pack_bytes` method from `uatypes.py`

oroulet, 2016-09-23:
ok. that was the missing information! yes you need to call ua.ua_binary.Primitives.Bytes.pack(xxxx)

Can you fix, test and send a pull request.

oroulet, 2016-09-23:
looks like we are missing for a test for password logging .....

closed 2016-10-05 by AlexejStukov