FreeOpcUa/python-opcua #309: TimeoutError

jaanraadik, 2016-10-04:
I recently migrated my collector script to a new server using an updated futures and python version, along with the latest build of python-opcua (before was using an August version). I'm using Intellution OPC software server side which hasn't been changed.

I'm commonly running into TimeoutErrors now similar to the one below:
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib64/python2.7/threading.py", line 811, in __bootstrap_inner
    self.run()
  File "../opcua/client/client.py", line 58, in run
    self.client.open_secure_channel(renew=True)
  File "../opcua/client/client.py", line 267, in open_secure_channel
    result = self.uaclient.open_secure_channel(params)
  File "../opcua/client/ua_client.py", line 222, in open_secure_channel
    return self._uasocket.open_secure_channel(params)
  File "../opcua/client/ua_client.py", line 163, in open_secure_channel
    response = ua.OpenSecureChannelResponse.from_binary(future.result(self.timeout))
  File "/usr/lib/python2.7/site-packages/concurrent/futures/_base.py", line 407, in result
    raise TimeoutError()
TimeoutError

No handlers could be found for logger "opcua.common.subscription"

I've seen a previous issue about Timeouts but this doesn't seem relevant. I've tried adding in a while loop limited to 10 occurrences around all places this exception is raised to retry where failure is observed to no avail

jaanraadik, 2016-10-04:
Also receiving:
'No handlers could be found for logger "opcua.client.ua_client.Socket" without any additional error information

cedricleroy, 2016-10-04:
I have exactly the same one. I first thought about network problems, but did test on my local machine where the OPC Server and the client are all together. It is kinda random and does not seem to be linked to loading (I did test 1000 subscription, 1 refresh per second). On my two days continuous testing, it did happen 5 times. 

I put in place an auto-reconnect with a watchdog, but would be definitely interested solving this one. I had a wireshark running during six hours today, but nothing did happen so far..

jaanraadik, 2016-10-04:
I have a poll thread running to restart the whole thing also - I'm subscribing to around 100 nodes, and usually during init of these one will fail with a TimeoutError. I now loop on these too, and often on the 2-4th loop it will successfully connect without a TimeoutError.

I thought perhaps it was an issue with the keepalive however I reduced this from the 1 hour default to intervals of 30 seconds to 30 minutes with no change.

I'm almost certain it's not an incompatibility with the server software I'm running as there was no issue until I deployed with updated versions of everything.

cedricleroy, 2016-10-04:
Which version did you run before upgrading? I am running 0.10.12 (yea, I should probably use the latest...). Didn't notify anything during init om my side, and in case it helps, I am using TopServer (Kepware).

zerox1212, 2016-10-04:
I'm not an expert on the binary protocol, but I am sure people will ask for a wire shark session to help debug.

oroulet, 2016-10-04:
It would be great if someone managed to find out when this started to
happen. You can also try to have a look at the log and see if we changed
something r related to timeout.
The logger error seems to come from the split of binary method from other
classes. I will have a look...

oroulet, 2016-10-04:
no it is not due to split...Would be great if you managed to nail it down further...

cedricleroy, 2016-10-04:
So, I have a way to reproduce this error systematically: I subscribe to 10000 nodes :) 

I have a wireshark log if interested. How can I send that out without sharing to the whole internet?

oroulet, 2016-10-04:
If you subscribe to 1000 nodes the log is probably impossible to analyse.
But if you have a quick way to reproduce the error and you think it was not
here before, you can test different versions , and find out when there bug
appeard. Even better clone git master and user git bisect to pick the exact
commit.

cedricleroy, 2016-10-04:
Ok, so I did run a couple of tests. Here is the code I used:

```python
from opcua import Client
import time
import Queue

q = Queue.Queue()

class OPCSubHandler(object):
    def datachange_notification(self, node, val, data):
        q.put(val)

client = Client("opc.tcp://%s:%s" % ("XXX.XXX.XXX.XXX", XXXXX))

client.connect()
handler = OPCSubHandler()
sub = client.create_subscription(1000, handler)

objects = client.get_objects_node()
s = set([objects])
nodelist = []

while s:
    node = s.pop()
    childs = node.get_children()
    tagname = str(node.nodeid.Identifier)
    if '_System' not in tagname and '_Statistics' not in tagname\
            and '_Comm' not in tagname and 'ObjectStatistics_' not in tagname:
        if not childs:
            if type(node.nodeid.Identifier) == str:
                nodelist.append(node)
        for child in childs:
            s.add(child)

# The below fail:
# sub.subscribe_data_change(nodelist)

for node in nodelist:
    sub.subscribe_data_change(node)

print "finish to subscribe"

while True:
    time.sleep(1)

client.disconnect()
```

Using python 2.7.6, Linux Mint (VM), TopServer (Kepware) on the host machine (Windows). 

The test routine is:
* Get all the nodes from the OPC Server (10 000)
* Subscribe to all of them
* Once subscription is finished, wait 10 minutes
* If no crash, the test pass

Results:
[image]

I was about using git bisect, but there is no guaranties that the error is between 0.10.2 and 0.10.3 ... The crash happening at 0.10.3 might be another bug introduced at that time.

oroulet, 2016-10-04:
Yes, the bug at 0).10.3 was something else, that got fixed. Looks like the timeout started at 0.10.10 if it was not hidden by the uaerror stuff.......

oroulet, 2016-10-04:
Did you try to increase the connection timeout in constructor of client? Maybe this is just an overload of server....

cedricleroy, 2016-10-04:
Just tried 0.10.12 with 10 seconds timeout and got the error. If it was a server overload, I suppose we will have the problem all the time whatever the version is.

cedricleroy, 2016-10-05:
Here is an update:
[image]

Pretty hard to figure out honestly. I am not sure if the Timeout we are seeing all along is not caused by something else. 

Looking at the code I was thinking at 10c38dccd926451d0060012855fd250febcd657d, but it looks it came with 0.10.7 only. 10c38dccd926451d0060012855fd250febcd657d had significant modifications to create_monitored_items.

zerox1212, 2016-10-05:
Can you test the same thing with Python 3.5?

cedricleroy, 2016-10-05:
Excellent idea. 

Did run 0.10.17, Python 3.5, subscribing to 10k tags during 20 minutes and no crash. 

Would the problem be linked to trollius?

oroulet, 2016-10-05:
So this is python 2.7 specific? Trollius is not maintained anymore. One day
we will need to remove support for python 2.7...

cedricleroy, 2016-10-05:
I didn't have a crash yet whereas I would probably have one with 2.7. But, due to the random nature of the problem, cannot be 100% sure yet. I am going to use 3.5 on our larger app and let it run a couple of days, and will confirm the hypothesis.

cedricleroy, 2016-10-06:
Bad news, I somehow have the same frequency of crashes. Did test with 200 nodes, and it did crash after ~ 55 minutes. What is bad also is that I cannot catch the error on my main application logs (does not raise any exception).

Here is what I have see on the shell though;

Protocol Error
Traceback (most recent call last):
  File "/home/project/venv3/lib/python3.5/site-packages/opcua/client/ua_client.py", line 94, in _run
    self._receive()
  File "/home/project/venv3/lib/python3.5/site-packages/opcua/client/ua_client.py", line 103, in _receive
    msg = self._connection.receive_from_socket(self._socket)
  File "/home/project/venv3/lib/python3.5/site-packages/opcua/ua/uaprotocol_hand.py", line 667, in receive_from_socket
    return self.receive_from_header_and_body(header, utils.Buffer(body))
  File "/home/project/venv3/lib/python3.5/site-packages/opcua/ua/uaprotocol_hand.py", line 639, in receive_from_header_and_body
    return self._receive(chunk)
  File "/home/project/venv3/lib/python3.5/site-packages/opcua/ua/uaprotocol_hand.py", line 670, in _receive
    self._check_incoming_chunk(msg)
  File "/home/project/venv3/lib/python3.5/site-packages/opcua/ua/uaprotocol_hand.py", line 596, in _check_incoming_chunk
    self.channel.SecurityToken.TokenId))
opcua.common.uaerrors._base.UaError: Wrong token id 75, expected 74
Protocol Error
Traceback (most recent call last):
  File "/home/project/venv3/lib/python3.5/site-packages/opcua/client/ua_client.py", line 94, in _run
    self._receive()
  File "/home/project/venv3/lib/python3.5/site-packages/opcua/client/ua_client.py", line 103, in _receive
    msg = self._connection.receive_from_socket(self._socket)
  File "/home/project/venv3/lib/python3.5/site-packages/opcua/ua/uaprotocol_hand.py", line 667, in receive_from_socket
    return self.receive_from_header_and_body(header, utils.Buffer(body))
  File "/home/project/venv3/lib/python3.5/site-packages/opcua/ua/uaprotocol_hand.py", line 639, in receive_from_header_and_body
    return self._receive(chunk)
  File "/home/project/venv3/lib/python3.5/site-packages/opcua/ua/uaprotocol_hand.py", line 670, in _receive
    self._check_incoming_chunk(msg)
  File "/home/project/venv3/lib/python3.5/site-packages/opcua/ua/uaprotocol_hand.py", line 620, in _check_incoming_chunk
    .format(self._peer_sequence_number, num))
opcua.common.uaerrors._base.UaError: Wrong sequence 19953 -> 19955 (server bug or replay attack)
Protocol Error
Traceback (most recent call last):
  File "/home/project/venv3/lib/python3.5/site-packages/opcua/client/ua_client.py", line 94, in _run
    self._receive()
  File "/home/project/venv3/lib/python3.5/site-packages/opcua/client/ua_client.py", line 103, in _receive
    msg = self._connection.receive_from_socket(self._socket)
  File "/home/project/venv3/lib/python3.5/site-packages/opcua/ua/uaprotocol_hand.py", line 667, in receive_from_socket
    return self.receive_from_header_and_body(header, utils.Buffer(body))
  File "/home/project/venv3/lib/python3.5/site-packages/opcua/ua/uaprotocol_hand.py", line 639, in receive_from_header_and_body
    return self._receive(chunk)
  File "/home/project/venv3/lib/python3.5/site-packages/opcua/ua/uaprotocol_hand.py", line 670, in _receive
    self._check_incoming_chunk(msg)
  File "/home/project/venv3/lib/python3.5/site-packages/opcua/ua/uaprotocol_hand.py", line 620, in _check_incoming_chunk
    .format(self._peer_sequence_number, num))
opcua.common.uaerrors._base.UaError: Wrong sequence 19953 -> 19956 (server bug or replay attack)
Exception in thread Thread-8:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/home/project/venv3/lib/python3.5/site-packages/opcua/client/client.py", line 59, in run
    val = server_state.get_value()
  File "/home/project/venv3/lib/python3.5/site-packages/opcua/common/node.py", line 129, in get_value
    result = self.get_data_value()
  File "/home/project/venv3/lib/python3.5/site-packages/opcua/common/node.py", line 138, in get_data_value
    return self.get_attribute(ua.AttributeIds.Value)
  File "/home/project/venv3/lib/python3.5/site-packages/opcua/common/node.py", line 244, in get_attribute
    result = self.server.read(params)
  File "/home/project/venv3/lib/python3.5/site-packages/opcua/client/ua_client.py", line 281, in read
    data = self._uasocket.send_request(request)
  File "/home/project/venv3/lib/python3.5/site-packages/opcua/client/ua_client.py", line 76, in send_request
    data = future.result(self.timeout)
  File "/usr/lib/python3.5/concurrent/futures/_base.py", line 407, in result
    raise TimeoutError()
concurrent.futures._base.TimeoutError

Looks like being similar to #295 ?

oroulet, 2016-10-06:
Interesting bug but it is going to be hard to debug... Not sure where to
start....

cedricleroy, 2016-10-06:
What is interesting is that as per #295 both occur with Kepware / Topserver  ... I wonder if the same error would happen with another server.

zerox1212, 2016-10-09:
I grabbed the latest and I have the same errors. I run python opcua server with my imported address space. Connecting with UA Expert causes timeouts every few seconds.

oroulet, 2016-10-09:
@zerox1212 you are now describing a completely different scenario. With a
python server and extern client. If you really have timeouts, you should
open a new bug report. And test with older versions, I haven't had any
timeout with server for a long time, but maybe we introduced a bug...

zerox1212, 2016-10-09:
Ok, I'm investigating. I'm working on a lot of things at once so maybe I created the problem.

zerox1212, 2016-10-09:
@cedricleroy Does your server have method nodes by chance?

cedricleroy, 2016-10-09:
@zerox1212 : No, I don't.

I was about to give an update on that thread: I put a lot of logs on my apps to try to figure out what could cause those Timeout. I found that the error was occurring when CPU near to 100%, or memory full. Did some tweak on the app to be more efficient, and it has been running during 30 hours without crashing (>10k nodes).

jk987, 2016-11-06:
@cedricleroy 

It really seems to be te same problem like #295 
TimeoutError is only consequence of "wrong token id" and "wrong sequence number".
You mention the error occurs when CPU is overloaded. In that case I guess it might happen the same situation like in my case with slow PLC connected with even slower RS232. In such a case the OpenSecureChannelResponse comes sooner than response of previous request (in my case the WriteResponse). But not so soon to have updated TokenId from OpenSecureChannelResponse when the WriteResponse is checked.

oroulet, 2020-02-17:
a branch to test for anyone interested: #962

Please tell me if it helps, and send me output if you get a crash again
@oroulet