FreeOpcUa/python-opcua #310: Datachange Source (Client vs. Server)

kevincolyar, 2016-10-04:
I'm looking for a way to differentiate between data changes that were made by a client connection and those which were made by the server itself.  

My application runs calculations on a 4s interval that updates the server's nodes with its latest data.  Those nodes can also be written to by the client and applied to the next calculations.  I'm looking to have my server update its nodes without triggering a bunch of datachange events, but at the same time know that a client is writting new data.

zerox1212, 2016-10-05:
I think you asked this before?

Your server needs to write directly to the address space dict or you need to change your concept.

kevincolyar, 2016-10-05:
I did but I must have deleted it.  I couldn't find the old issue.

Would you guys frown on submitting a pull request to add a `with_callbacks` kwarg flag to the server's `write` method and pass it all the way down to the `AddressSpace#set_attribute_value`?  That way I don't have to duplicate everything in `set_attribute_value`.

Basically:

In InternalServer:

```python
    def write(self, params, with_callbacks=True):
```

In AddressSpace#set_attribute_value:

```python
    if with_callbacks and old.Value != value.Value:  # only send call callback when a value change has happend
```

kevincolyar, 2016-10-07:
I've been able to directly update the address space nodes and can see the changes in a simple python-opcua client.  However, my TOP Server (https://www.softwaretoolbox.com/topserver/) does not see updates.  The TOP Server does see the changes if I update node normally.  

Here is how I'm updating the node values:

```python
        val = ua.DataValue(ua.Variant(float(v), ua.VariantType.Float))
        val.SourceTimestamp = datetime.datetime.utcnow()
        val.ServerTimestamp = datetime.datetime.utcnow()
        val.StatusCode = ua.StatusCode(ua.StatusCodes.Good)
        address_space._nodes[nodeid].attributes[ua.AttributeIds.Value].value = val
```

Do I need to call something to trigger update events?

oroulet, 2016-10-08:
In python-opcua there is no difference between an external and internal
client. The subscription code is already quite complex so you would need a
very good case to add even more complexity by differentiating client
origins.
From your description of you application it sounds like you should have two
nodes: one read only updated internally and one for external clients that
is subscribed to internally and copy data to the read only mode.
But as I said if you have a very good use case we may think about it....

zerox1212, 2016-10-08:
I gave him the same advice when he asked this before.
https://github.com/FreeOpcUa/python-opcua/issues/145

bitkeeper, 2016-10-10:
In #145 a case was requested. Which approach is suggested in this case ?

**Example case for a typically communication bridge**

Node is a rw node for the client.

**Required functionality:**

[image]
- When the client sets the node value MyHandler should be called  (write_by_client)
- When MyHandler sets the value the client should is called (write_internally)

**The problem:**
For monitoring the value of the node also internally a subscription is used/
When MyHandler sets the value of Node, the subscription is also calles MyHandler.

Its makes no sense to make two nodes (one for rd and one for rw) in this case.

**Workarround:**
To prevent unnecessary updates store the latest value also in MyHandler. On `datachange_notification` compare if the value is changed. Downside is that a datavalues need to be duplicated.