FreeOpcUa/python-opcua #315: Dead historizing methods in server

zerox1212, 2016-10-06:
When I was working on some other things I saw that this is still in server.py:
```python
    def historize_node(self, node):
        self.iserver.enable_history(node)

    def dehistorize_node(self, node):
        self.iserver.disable_history(node)
```

These don't work, and in the examples historizing is done by calling `iserver` methods directly. Should we delete these or fix them to reference `iserver`?

destogl, 2016-10-06:
It would be the best to fix to reference correct methods and also that we have tests for them...

zerox1212, 2016-10-06:
If we fix them it's only for user convenience. The server method is just going to call the iserver method with the same parameters.

If I get time I will fix it and update tests.

zerox1212, 2016-11-15:
Fixed in https://github.com/FreeOpcUa/python-opcua/pull/353

closed 2016-11-15 by zerox1212