FreeOpcUa/python-opcua #317: Timeout Error with Imported Methods

zerox1212, 2016-10-09:
This one has me very confused.
1. I make a very basic address space in OPCUA Modeler, since it's running a server in the background I connect to it via UA Expert.
Modeler:
[image]
UA Expert connected:
[image]

This has no issues, clicking on the Input or Output Arguments returns node information as expected. I can even call the method (even though it just returns nothing to do).

I then export these 3 nodes to XML to method.xml. Then I start a server and call `server.import_xml("method.xml")`. The importer migrates everything to namespace 3, which should be fine. I connect UA Expert and I can browse the object and method nodes with no problems.
Browsing upper nodes with UA Expert:
[image]

However, as soon as I click on the InputArgument or OutputArgument nodes, the server crashes with this:
```
ERROR:opcua.server.binary_server_asyncio:Exception raised while parsing message from client, closing
Traceback (most recent call last):
  File "..\opcua\server\binary_server_asyncio.py", line 85, in _process_data
    ret = self.processor.process(hdr, buf)
  File "..\opcua\server\uaprocessor.py", line 87, in process
    return self.process_message(msg.SecurityHeader(), msg.SequenceHeader(), msg.body())
  File "..\opcua\server\uaprocessor.py", line 108, in process_message
    return self._process_message(typeid, requesthdr, algohdr, seqhdr, body)
  File "..\opcua\server\uaprocessor.py", line 184, in _process_message
    self.send_response(requesthdr.RequestHandle, algohdr, seqhdr, response)
  File "..\opcua\server\uaprocessor.py", line 41, in send_response
    response.to_binary(), message_type=msgtype, request_id=seqhdr.RequestId, algohdr=algohdr)
  File "..\opcua\ua\uaprotocol_auto.py", line 8933, in to_binary
    packet.append(fieldname.to_binary())
  File "..\opcua\ua\uatypes.py", line 996, in to_binary
    packet.append(self.Value.to_binary())
  File "..\opcua\ua\uatypes.py", line 837, in to_binary
    b.append(uabin.pack_uatype_array(self.VariantType, flatten(self.Value)))
  File "..\opcua\ua\ua_binary.py", line 220, in pack_uatype_array
    b = [pack_uatype(vtype, val) for val in array]
  File "..\opcua\ua\ua_binary.py", line 220, in <listcomp>
    b = [pack_uatype(vtype, val) for val in array]
  File "..\opcua\ua\ua_binary.py", line 235, in pack_uatype
    return extensionobject_to_binary(value)
  File "..\opcua\ua\uaprotocol_auto.py", line 16339, in extensionobject_to_binary
    Body = obj.to_binary()
  File "..\opcua\ua\uaprotocol_auto.py", line 951, in to_binary
    packet.append(self.DataType.to_binary())
  File "..\opcua\ua\uatypes.py", line 405, in to_binary
    return struct.pack("<BB", self.NodeIdType.value, self.Identifier)
struct.error: required argument is not an integer
ERROR:opcua.server.binary_server_asyncio:Exception raised while parsing message from client, closing
Traceback (most recent call last):
  File "..\opcua\server\binary_server_asyncio.py", line 85, in _process_data
    ret = self.processor.process(hdr, buf)
  File "..\opcua\server\uaprocessor.py", line 77, in process
    msg = self._connection.receive_from_header_and_body(header, body)
  File "..\opcua\ua\uaprotocol_hand.py", line 639, in receive_from_header_and_body
    return self._receive(chunk)
  File "..\opcua\ua\uaprotocol_hand.py", line 670, in _receive
    self._check_incoming_chunk(msg)
  File "..\opcua\ua\uaprotocol_hand.py", line 620, in _check_incoming_chunk
    .format(self._peer_sequence_number, num))
opcua.common.uaerrors._base.UaError: Wrong sequence 81 -> 83 (server bug or replay attack)
```

Any ideas?

zerox1212, 2016-10-09:
I think I found the root cause, but i'm not sure how to solve it yet...
During import of extension object the code creates a TwoByteNodeId, then assigns it a NumericNodeId:
[image]

zerox1212, 2016-10-13:
This is now solved, but it should be mentioned that it appears if *any* binary issue occures it throws `opcua.common.uaerrors._base.UaError: Wrong sequence 81 -> 83 (server bug or replay attack)` . This seems like a side affect of refactoring ua errors.

closed 2016-10-13 by zerox1212

oroulet, 2016-10-13:
That is a very interesting information @zerox2121 !!! Can you open a bug
report?