FreeOpcUa/python-opcua #321: command line test tool for OPC-UA method call missing in python-opcua

SCHUNK-Osswald, 2016-10-10:
Changing values in our (proprietary) opc-ua server is done via "OPC-UA method calls". For simple test it would thus be helpful to have a "uacall" command line tool, just like "uaread/uawrite/...". 
The rest of the code seems to be prepared for providing this (except for a minor bug in opcua.node.Node.call_method (missing self parameter)).

I feel sufficiently skilled to add this myself, so will try to provide some coding help here. (But please be forgiving, as I'm new to git/github/opcua)

oroulet, 2016-10-10:
Sounds like a good idea. It should not be difficult..

zerox1212, 2016-10-10:
If you don't want to learn git, just grab the github client. It works well.

The only thing that sucks about the gui client is it's often hard to synchronize you fork with the original master. In which case I open the GitShell and use the following command:

```
git remote add upstream url/to/original/repo
git fetch upstream
git checkout master
git reset --hard upstream/master
git push origin master --force
```

closed 2016-11-02 by oroulet