FreeOpcUa/python-opcua #324: fix namespace import

oroulet, 2016-10-11:
@bitkeeper @maljac 
1) move namespace index collision code out of xmlparser. it should be in xmlimporter
2) if commit https://github.com/FreeOpcUa/python-opcua/commit/1a5475f182d600050ffa11d392bded1844808a05 is uncommited, you will see that we happily register again the same namespace. So this commit should be remove and a proper fixed implemented

to reproduce: 
1- uncommit https://github.com/FreeOpcUa/python-opcua/commit/1a5475f182d600050ffa11d392bded1844808a05
2- in tests/tests_xml add following at beiynning of test_xml_export
self.opc.register_namespace("whatever")
self.opc.register_namespace("whatever2")
run test:
python tests/tests.py TestServer.test_xml_export
it will fail

bitkeeper, 2016-10-11:
I have reproduced the problem as you described it. 
Maybe the xmlimport isn't the problem, but the xmlexport:

My reference here is UaModeler. I use it with multiple namespace including import/export in UaModeller and import only on the python-opcua side. In UaModeler I can only use namespace indexes of namespace that are registered. Index 0 is always present.

If I make,in UaModeler, a model with 5 namespaces and lets say export namespace 4&5 they are written in the xml as ns 1&2. That has to do with the `NamespaceUris` XML Element. It is an array not a dictionary:
```xml
  <NamespaceUris>
    <Uri>whatever</Uri>
    <Uri>whatever2</Uri>
  </NamespaceUris>
```
All ns of the node ids and browse are changed from 4&5 to 1&2. Lets call it relative namespaces indexes, based on namespace array in the xml file.

The current implementation of python-opcua `xml_import` does the opposite. On import it registers the namespaces (registering an already existing namespace isn't a problem, it just returns the existing) from the xml file and on import of the nodeis replace it with the correct ns*<sup>1</sup> index.
If there are there isn't an index present in the file, it leaves the ns indexes alone.

In the ends UaModeler and xml import of python-opcua works nice together; both use the same strategy. The only thing is that I can't control the namespace index of certain namespace, than by  preregister namespace in a certain order...

The `xml_export` of python-opcua is something different. It also adds  `NamespaceUris` XML Element to the xml file, but it doesn't  make the nodeid relative to the namespace array in the file. 
And that is where the problem starts on the import *<sup>2</sup>!

@oroulet @maljac 
What is your view on it?

\*<sup>1</sup> That is why the `XmlParser.__iter__` uses `self.namespaces[ns_index + 1]`.
\*<sup>2</sup> Btw if the export doesn't write the `NamespaceUris` the import works just fine, which makes sense because the import now uses the unchanged nodeids.

oroulet, 2016-10-11:
Thanks for the analyses. sounds correct. In uamodeller can you choose which namespace to export? I am wondering if we should always export all namespaces or exported selected namespaces (as we currently do) or simply export namespaces needed by the nodes we export.

bitkeeper, 2016-10-11:
Default  uamodeler export only the selected namespace . 
The index of the export namespace is always 1 in the xml document. Referenced namespaces are always 2 or higher and have also an entry in the namespacearray expect for the default namespace 0.

You can also export multiple namespaces at once (menu > settings -> last option). Then you have to select which ones are included.

oroulet, 2016-10-11:
What happens if you export many nodes with idx 2 and 3 but only export namespace X with idx 4? We then get an xml file with one namespace X that might be imported as idx 2 thus some nodes would reference the wrong namespace. In my head the only correct solution is to always export namespaces if a node uses it

oroulet, 2016-10-12:
fix export: https://github.com/FreeOpcUa/python-opcua/pull/326
need to improve test

bitkeeper, 2016-10-12:
@oroulet 
Hereby an example of example exports, I hope this answers your questions.

The model exists out of:

- namespace http://yourorganisation.org/types
- namespace http://yourorganisation.org/instances
- objectype MyType in namespace http://yourorganisation.org/types
- objece MyObject in namespace http://yourorganisation.org/instances

The following exports are provided below:
- export of namespace  types + instances in one file
- export of namespaces types only
- export of namespaces instance only
Especially the latest in interesting because MyObject refers to an MyType in the other namespace.

Export of namespace  types + instances in one file:
```xml
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:s1="http://yourorganisation.org/types/Types.xsd" xmlns:s2="http://yourorganisation.org/instances/Types.xsd" xmlns:ua="http://unifiedautomation.com/Configuration/NodeSet.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema">
    <NamespaceUris>
        <Uri>http://yourorganisation.org/types/</Uri>
        <Uri>http://yourorganisation.org/instances/</Uri>
    </NamespaceUris>
    <Aliases>
        <Alias Alias="Organizes">i=35</Alias>
        <Alias Alias="HasModellingRule">i=37</Alias>
        <Alias Alias="HasTypeDefinition">i=40</Alias>
        <Alias Alias="HasSubtype">i=45</Alias>
        <Alias Alias="HasComponent">i=47</Alias>
    </Aliases>
    <UAObjectType NodeId="ns=1;i=1002" BrowseName="1:MyType">
        <DisplayName>MyType</DisplayName>
        <References>
            <Reference ReferenceType="HasComponent">ns=1;i=7001</Reference>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=58</Reference>
        </References>
    </UAObjectType>
    <UAMethod ParentNodeId="ns=1;i=1002" NodeId="ns=1;i=7001" BrowseName="1:myMethod">
        <DisplayName>myMethod</DisplayName>
        <References>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=1002</Reference>
        </References>
    </UAMethod>
    <UAObject NodeId="ns=2;i=5002" BrowseName="2:MyObject">
        <DisplayName>MyObject</DisplayName>
        <References>
            <Reference ReferenceType="HasComponent">ns=2;i=7001</Reference>
            <Reference ReferenceType="Organizes" IsForward="false">i=85</Reference>
            <Reference ReferenceType="HasTypeDefinition">ns=1;i=1002</Reference>
        </References>
    </UAObject>
    <UAMethod ParentNodeId="ns=2;i=5002" NodeId="ns=2;i=7001" BrowseName="1:myMethod" MethodDeclarationId="ns=1;i=7001">
        <DisplayName>myMethod</DisplayName>
        <References>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=2;i=5002</Reference>
        </References>
    </UAMethod>
</UANodeSet>
```
Export of namespaces types only:
```xml
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:s1="http://yourorganisation.org/types/Types.xsd" xmlns:ua="http://unifiedautomation.com/Configuration/NodeSet.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema">
    <NamespaceUris>
        <Uri>http://yourorganisation.org/types/</Uri>
    </NamespaceUris>
    <Aliases>
        <Alias Alias="HasModellingRule">i=37</Alias>
        <Alias Alias="HasSubtype">i=45</Alias>
        <Alias Alias="HasComponent">i=47</Alias>
    </Aliases>
    <UAObjectType NodeId="ns=1;i=1002" BrowseName="1:MyType">
        <DisplayName>MyType</DisplayName>
        <References>
            <Reference ReferenceType="HasComponent">ns=1;i=7001</Reference>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=58</Reference>
        </References>
    </UAObjectType>
    <UAMethod ParentNodeId="ns=1;i=1002" NodeId="ns=1;i=7001" BrowseName="1:myMethod">
        <DisplayName>myMethod</DisplayName>
        <References>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=1002</Reference>
        </References>
    </UAMethod>
</UANodeSet>
```

Export of namespaces instance only (but uses the type ns):
```xml
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:s1="http://yourorganisation.org/instances/Types.xsd" xmlns:s2="http://yourorganisation.org/types/Types.xsd" xmlns:ua="http://unifiedautomation.com/Configuration/NodeSet.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema">
    <NamespaceUris>
        <Uri>http://yourorganisation.org/instances/</Uri>
        <Uri>http://yourorganisation.org/types/</Uri>
    </NamespaceUris>
    <Aliases>
        <Alias Alias="Organizes">i=35</Alias>
        <Alias Alias="HasTypeDefinition">i=40</Alias>
        <Alias Alias="HasComponent">i=47</Alias>
    </Aliases>
    <UAObject NodeId="ns=1;i=5002" BrowseName="1:MyObject">
        <DisplayName>MyObject</DisplayName>
        <References>
            <Reference ReferenceType="HasComponent">ns=1;i=7001</Reference>
            <Reference ReferenceType="Organizes" IsForward="false">i=85</Reference>
            <Reference ReferenceType="HasTypeDefinition">ns=2;i=1002</Reference>
        </References>
    </UAObject>
    <UAMethod ParentNodeId="ns=1;i=5002" NodeId="ns=1;i=7001" BrowseName="2:myMethod" MethodDeclarationId="ns=2;i=7001">
        <DisplayName>myMethod</DisplayName>
        <References>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=5002</Reference>
        </References>
    </UAMethod>
</UANodeSet>
```

oroulet, 2016-10-12:
So that seems to mean that namespaces are always exported if they are used.
Both as type, instance or type.

oroulet, 2016-10-13:
https://github.com/FreeOpcUa/python-opcua/pull/327
btw it looks like we should also export namespace if they are udes by browsenames, not only by node ids..... will add it

oroulet, 2016-10-13:
export of ns in browse_names done

zerox1212, 2016-10-13:
I'm not totally sure if your description is always true @bitkeeper . You can make an address space where types and instances are both in NS 1. What you have seems more like convention than a hard rule.

bitkeeper, 2016-10-13:
@zerox1212 I think you misunderstood me; I'm not claiming that instances and types should be separated namespaces. It is just an example of two namespaces with a dependency between them and how it  looks if you make export of those namespaces. It demonstrates howto deal with relative namespace indexes on an import/export.

zerox1212, 2016-10-21:
I don't really understand the PR. Is this issue now solved?

oroulet, 2016-10-22:
Should be fixed in  https://github.com/FreeOpcUa/python-opcua/commit/136b6fd08aa27aa3cf7b2e1f52b186bf3b7306f9

@zerox1212 this changes quite a lot how namespaces are exported. You should test

zerox1212, 2016-10-24:
I will grab latest master soon and test. Was kind of waiting for the default value code to be merged because it's required for my application.

oroulet, 2016-10-25:
Supposed to be solved by my last commit. I also added tests.

zerox1212, 2016-10-26:
I have been working with the latest master. My application doing imports and exports seems to work good. Can probably close this. If anything comes up a new issue can be made.

closed 2016-11-02 by oroulet