FreeOpcUa/python-opcua #332: DataChangeFilter with trigger to timestamp.

brubbel, 2016-10-15:
Hi,
I'm fairly new to OPC UA and was looking into installing a custom DataChangeFilter to receive data even when the value has not changed.
```
Simple example:
        while True:
            time.sleep(1)
            count += 0.1
            myvar.set_value(round(count))
```
However, looking at the code in address_space.py, it seems that identical values are already pre-filtered before they even reach the datachange_callbacks():
```
if old.Value != value.Value:  # only send call callback when a value change has happend
    cbs = list(attval.datachange_callbacks.items())
```
So, could it be that installing a filter which triggers on a timestamp or a zero deadband will fail in advance?

oroulet, 2016-10-16:
will fail. neither the spec not the code support such a thing. what are you trying to achieve?

brubbel, 2016-10-16:
Ok, I see that opc does not support repeated values. 
My use case is a wireless sensor which sends regular 
updates, which could have the same value. The way opc works masks the difference between a connection loss and a static value, but I guess the appropriate solution will be to combine a 'last seen' value along with the measurement.
However, this somewhat defeats the purpose of minimizing traffic.

oroulet, 2016-10-16:
The opcua foundation is now working on a real-time extension to opcua that
is not real-time at all since based on UDP/amqp but it will solve that kind
of cases. But someone has to implement this in python.... Or pay someone to
do it..

oroulet, 2016-10-16:
Btw You can use events. That will work fine, and you can discover lost
events and do history query

brubbel, 2016-10-16:
Thanks! I think this issue can be closed.
I may append some info later on, when I have a better view on opc-ua with a few alternative options as a reference for others with the same question.

closed 2016-10-16 by brubbel

zerox1212, 2016-10-16:
FYI

If your wireless sensor is where the OPC UA server lives, you should be able to just call node.set_value() with the same value and it will trigger client updates. In your example you can test it easily be seeing if the node's timestamp is changing every second.