FreeOpcUa/python-opcua #333: Move Some User Values Out of server.py

zerox1212, 2016-10-17:
I think we should refactor server.py and all of the examples to move these uri/namespace registration outside of the server code.

```python
 self.endpoint = urlparse("opc.tcp://0.0.0.0:4840/freeopcua/server/")
        self.application_uri = "urn:freeopcua:python:server"
        self.product_uri = "urn:freeopcua.github.no:python:server"
        self.name = "FreeOpcUa Python Server"

# setup some expected values
        self.register_namespace(self.application_uri)
        sa_node = self.get_node(ua.NodeId(ua.ObjectIds.Server_ServerArray))
        sa_node.set_value([self.application_uri])
```

It will be more clear for new users, especially when importing an address space from XML.

Thoughts?

oroulet, 2016-10-18:
The spec requires this values. These attributes are documented in server class and can be overriden by users if they want, isn'it? Every other SDK I have seen also give some default values to these attributes. Why are they disturbing you?

zerox1212, 2016-10-18:
When I use modeler I put all my stuff in NS1 for example. Then I write my python user application. In my application I might do something like this:
`machine_node = self.server.nodes.objects.get_child("1:Machine0")`

Now when I run my application and import the xml, Machine0 will end up in NS2 and my application no longer works.

However I suppose the correct solution is to only create my address space in NS2 or higher, but I just find it kind of confusing. Especially if you running the server library from a pip installation.

oroulet, 2016-10-18:
The behaviour  make sense. Idx 1 is always used by server. The correct
implementation is to not rely on hard coded  idx but use get_ namespace _
index at startup to get correct idx

zerox1212, 2016-10-18:
Ok, thanks.

closed 2016-10-18 by zerox1212